google/xls #556: MLIR integration

makslevental, 2022-01-20:
Do you guys have any interest in integrating with MLIR? I.e., supporting (morally even if not materially) an MLIR [dialect](https://mlir.llvm.org/docs/Dialects/) corresponding to XLS IR. The valueprop would be that it could enable lowering from the various MLIR dialects (modulo those legalizations being implemented) to XLS. High-level use case would look something like lowering PyTorch to XLS (through the [Torch MLIR dialect](https://github.com/llvm/torch-mlir)).

cdleary, 2022-01-20:
Hi Maksim, that's not something we'd pursue ourselves, and we don't want to take MLIR as a dependency in this particular project repo, but if there's a project/effort to attempt to integrate things like that we'd be happy to cite it in our documentation / as a related project/repo that's using XLS.

One thing I'd caution just by a contrast is that while _XLA_ is a bit more stable (there's a bunch of different dialects that attempt to project XLA HLO into MLIR is my understanding, with various tradeoffs / attempted extensions), _XLS_ is still very fast moving, and our fundamental primitives/approach is more subject to change over time, so being hooked into XLS development and direction will likely be necessary to know how to effective "rebase" the integration onto the latest-and-greatest capabilities as the project evolves.

Hope that helps clarify, thanks for the question!

makslevental, 2022-01-21:
@cdleary thanks for the response.

I'm not 100% sure (since I'm a nOOb when it comes to both the compiler side and the synthesis side) but I don't think any such "integration" would actually entail taking on MLIR as a dependency. XLS IR would be downstream of the MLIR dialect so in the reality XLS would be the dependency (but only weakly). That's why I phrased it as "moral" support, along the lines of open lines of communication about BC. 

Anyway thanks for working on XLS in the open - I think it's a great example of how to build HLS from scratch.

closed 2022-01-21 by makslevental

cdleary, 2022-01-21:
Makes sense, always happy to try to communicate about use cases! I'm not sure what "BC" refers to there, but we are doing XLS open source to enable all sorts of collaboration and use cases, so greatly appreciate the feedback, and feel free to reach out again.