google/xls #559: Use range analysis to simplify array_index operations

meheff, 2022-01-22:
If the range of the index value of an array index operation is limited, then some simplification may be possible:

1.  all indexed elements within the possible range of the index are identical. In this case the array_index operation can be replaced with that element.
2.  the indices within the range index very few different values relative to the size of the array. In this case, the array_index can be replaced with a select between the possible elements.

An internal benchmark showed a couple instances of this:

* Indexing into a literal array with all identical elements.
* Indexing into a literal array with different elements but where the index has only one non-constant bit.

The benchmark suggests that you may be able to get good mileage out of something simple to start. For example, do case (1) above and for case (2) only do the transform when selecting between two elements and where the selector can be trivially expressed (say the range is only two values).  This would be easy to generalize over time.

meheff, 2022-01-22:
@taktoa fyi

meheff, 2022-01-22:
Another interesting case (also found in the benchmark) is where one of the indices is a literal. Example:

```
A = [[a, b, c], [d, e, f], [g, h, i]]
result = array_index(A, {x, 1})
```

In this case we can transform this to:

```
A' = [b, e, h]
result = array_index(A, {x})
```

Range analysis is not strictly necessary here, but this transformation is of the same ilk.

taktoa, 2022-01-22:
Interesting, I'll definitely take a look

commit 51597c9d (2022-02-15): Add narrowing pass that replaces ArrayIndex with a Select chain

Closes #559

PiperOrigin-RevId: 428665359
closed 2022-02-15 by copybara-service[bot]