google/xls #575: Allow for user-specified input/output delay

meheff, 2022-02-13:
For pipelined designs it is sometimes desirable to allow inputs to flow through some amount of combinational logic before being flopped. The amount of delay allowed through this logic would be user specified. A typical setting might be 50% of the clock period. Symmetrically, outputs could support this feature as well.

A simple, but imprecise way of handling this is to give receive nodes some amount of delay. Problems with this approach (from @hongted ):

> Couple of limitations are
> 1) it doesn't exactly place an additional delay on the input port -- but rather the receive op possibly resulting in double counting if those receive ops are chained via tokens>
> 2) even if receive ops aren't chained by tokens, one receive op could feed the predicate of the other -- again resulting in double counting.

My (@meheff ) response:

> (1) could be resolved by fanning out tokens to parallel receives rather than serializing them. This would naturally be what we would do if a token dependency implies separate cycles for the dependent ops.
> 
> (2) for this we'd need something more sophisticated. Essentially what we want is the receive to be scheduled no earlier than N ps in the cycle. This could be enforced via a special rule. Seems like it could work but I don't know all the ramifications of this.

For output delay, I think just given sends the desired delay should work because there can be no data path between sends. I.e, (2) is not a problem.

hongted, 2024-03-06:
Adding in comment by @cdleary  from https://github.com/google/xls/issues/1335

I imagine this can be wired up to e.g. param nodes, reflecting them having delay in their arrival, in the SDC scheduling problem.

This is a thing people do try to avoid for their blocks (vs a cleanly timed flopped path), but it happens, and I think it's not too burdensome to support.

My napkin sketch is along the lines of (for simple pipelines):

```
fn f(x: MyInputStruct) -> MyOutputStruct {
  ...
}
```

`codegen_main --param_delay=x:10ps`

(And of course `sched_main` would really be the ideal binary in the command line but #724 is not done)

allight, 2026-11-14:
Closing this issue as fixed (accepted by @allight): User-specified input and output delays are implemented via `SchedulingOptions::additional_input_delay_ps`, `SchedulingOptions::additional_output_delay_ps`, and per-channel `SchedulingOptions::additional_channel_delay_ps` in `xls/scheduling/scheduling_options.h:543-578` and enforced in `xls/scheduling/pipeline_schedule.cc` (tested in `xls/scheduling/pipeline_schedule_test.cc:1885-2025`). (Closed by Gemini)

closed 2026-11-14 by allight