google/xls #587: Enable I/O in DSLX modules

vincent-mirian-google, 2022-03-14:
Create a library that enables file I/O in DSLX modules.

Background: many co-design tools have this feature. The feature is typically used for tests. For XLS to be competitive, this feature would be nice.

This is a thought/enhancement (future work).

vincent-mirian-google, 2022-03-14:
Example with syntax (the syntax is only for demonstration purpose and may change):

```
import iostd;

fn main(a: u32) -> u32 {
  a + 1
}

#![test]
fn main_test() {
  let file = open(filepath);
  # loop from 0 until num lines within the file
  # note omitting line format.
  for (i, input_value, golden_value): (u32, u32, u32) in range(u32:0, u32:file::num_lines()) {
     let _ = assert_eq(main(input_value), golden_value);
  }
  ()
}
```

ted-xie, 2022-03-14:
Any system or file I/O would not be synthesizable - as long as this is clear to the end user then this feature could be useful for testing purposes.

If we're talking about synthesizable I/O, IMO there is a greater need for some kind of handshake interface so that XLS modules can integrate more easily in other chip projects. Something like [AXI Stream](https://zipcpu.com/doc/axi-stream.pdf) would be nice. This would be a game-changing feature and would help XLS inch closer to feature parity with competing HLS DSLs.

ted-xie, 2022-03-14:
Ultimately a coding style like this would be nice:

```
fn mymodule(a: u32_channel) -> u32) {
  let a_in = channel_handshake_read(a);
  a_in
}
```

vincent-mirian-google, 2022-03-15:
Hi Ted.

The idea presented in this thread was for testing purposes _only_.

Regarding synthesizable I/O, I agree the concept/construct of an interface attached to the a parameter of a DSLX function/proc would enable integration of the IP into a larger design.

vincent-mirian-google, 2022-03-15:
@ted-xie I added a separate issue (https://github.com/google/xls/issues/588) to describe the idea. You are welcome to add more thoughts to that thread. Thanks! :-)

cdleary, 2022-05-10:
Wanted to give an update for this issue: I don't think we're likely to do this with our current outlook (though things can always change). Currently the DSL test constructs are entirely synthesizable, and we rely on external Python or C++ testbenches (within the confines of Google-used languages, the latter for fast many-sampled tests) -- this decision forces us into a multi-language reality for non-synthesizable testing infrastructure (when using the DSL) but I think that has been reasonable for our purposes so far. In particular going forward we likely want to make sure we have good examples and reusable components for driving common design interfaces with minimal boilerplate. We have some examples in things like the quickcheck attribute and perhaps we could create other common attribute-like annotations, even tools packaged up as Bazel genrules that generate "DUT drivers" automatically seem like they'd be helpful.

Automatically stitching/packing a proc/fn interface to be exposed over something like an AXI stream does seem like something we continue to want (in the NoC sub-project I believe we usually used the terminology "interface blocks"), so I think along the lines of #588 and auto-wrapping / stitching is more of the future here, I'm compelled to close this one out given the rationale above but feel free to holler/reopen if there's additional points that were not addressed.

closed 2022-05-10 by cdleary