google/xls #588: Enable HW interface specification in DSLX modules

vincent-mirian-google, 2022-03-15:
Motivated by the discussion from https://github.com/google/xls/issues/587 with @ted-xie.

Enable HW interface specification in DSLX modules: The interface specification of a parameter in a DSLX function or proc would generate an interface in the hardware description of the IP to interface with other IP when integrating the IP within a larger design.

ted-xie, 2022-04-19:
Quick thought dump: really what needs to happen is that XLS-generated RTL should have the ability to flow-control the input and output interfaces. A simple ready/valid handshake with data should suffice, such as AXI-Stream with its TVALID/TREADY/TDATA. Having this type of flow-controllable interface is really crucial to enable XLS to be integrated in larger SoC projects.

cdleary, 2022-05-10:
Also notes relates to discussion in #578 -- "easy stitching of XLS components into pre-existing external-world environments" is something we can make more streamlined out of the box, given the places we've embedded in designs had dedicated integration bandwidth we didn't need it to be as "drag and drop" historically but this would clearly be enabling for more use cases.

allight, 2026-11-14:
Keeping this issue open (reviewed by @allight): While external SystemVerilog types via `#[sv_type("...")]` and `PackageInterfaceProto` were added (`xls/dslx/frontend/ast.cc:111`, `xls/codegen/clone_nodes_into_block_handler.cc:747-757`), we likely want more hardware interface specification support in this vein, so we are not closing this issue. (Summarized by Gemini)