google/xls #590: document `top` flag requirement for tools

proppy, 2022-03-18:
Following the https://google.github.io/xls/tools_quick_start/ yield the following error on ir optimization:
```
(base) rgb_mixer 🍡 opt_main src/simple_add.ir  > src/simple_add.opt.ir
opt_main src/simple_add.ir  > src/simple_add.opt.ir
F0319 02:08:55.584117   34528 opt_main.cc:115] Check failed: ::absl::OkStatus() == (xls::tools::RealMain(positional_arguments[0])) (OK vs. INTERNAL: Top entity not set for package: simple_add.) 
```

@meheffernan mentioned that xls now require to pass `--top` explicitly to `opt_main` and `codegen_main` and/or `--entry` to `ir_conversion_main`.

We should make sure this is documented everywhere the documentation refer to the invocation of those tools:
https://github.com/google/xls/search?q=path%3Adocs_src+opt_main&type=code
https://github.com/google/xls/search?q=path%3Adocs_src+codegen_main&type=code
https://github.com/google/xls/search?q=path%3Adocs_src+ir_conversion_main&type=code

cdleary, 2022-04-26:
@vincent-mirian-google Saw there was a bunch of activity towards this, resolved now / can we close?

vincent-mirian-google, 2022-04-26:
The issue is resolved. https://google.github.io/xls/tools_quick_start/ references the 'top' argument.

closed 2022-04-26 by vincent-mirian-google