google/xls #606: Pass that turns data dependencies into token dependencies

taktoa, 2022-04-12:
For proc inlining and the upcoming mutual exclusion pass, we'll need a pass that turns data dependencies between effectful operations into token dependencies. A sketch of how this might work:

- Compute a relation `D` that is the transitive closure of the data dependency graph
- Run the token provenance pass to get a relation `T` that is the token dependency graph
- Iterate over all effectful nodes that produce data (currently only receives), call that node `a`
  - Iterate over all effectful descendants of `a` in the data dependency graph (can be determined with `D`), call that node `b`
    - Replace the token input of `b` with `AfterAll(token_output_of_a, old_token_input_of_b)`

cdleary, 2022-04-26:
@taktoa Can you add an example before / after program to serve as an example here of the analysis / transformation?

taktoa, 2022-04-26:
Sure, here's an example:

```
(token_B, data) = receive(token_A, ...)
token_C = send(token_A, data, ...)
token_D = after_all(token_B, token_C)
next (token_D, ())
```

would become (minimally)

```
(token_B, data) = receive(token_A, ...)
token_C = send(token_B, data, ...)
token_D = after_all(token_B, token_C)
next (token_D, ())
```

The sketch I gave would instead generate

```
(token_B, data) = receive(token_A, ...)
token_send = after_all(token_A, token_B)
token_C = send(token_send, data, ...)
token_D = after_all(token_B, token_C)
next (token_D, ())
```

which is of course semantically equivalent to the second example.

The output of this pass does prevent some optimizations. For example, if you had something like

```
(token_B, data) = receive(token_A, ...)
actually_zero = and(0, data)
token_C = send(token_A, actually_zero, ...)
token_D = after_all(token_B, token_C)
next (token_D, ())
```

then an optimization could sever the data dependency between the receive and the send, thus eliminating the happens-before relation, but after this pass runs the happens-before relation is now part of the token network and cannot be removed by optimizations. In practice, this can be mitigated in most cases by only running this pass after at least a round of simplification passes.

commit 7e031dc4 (2022-05-06): Add token dependency pass

This pass adds token dependencies between side effectful operations
that have data dependencies between them. This is conservative, since
it may be possible to eliminate the data dependency by optimizations,
but it is useful for making the proc inlining and mutual exclusion
passes work.

Closes #606

PiperOrigin-RevId: 447039108
closed 2022-05-06 by copybara-service[bot]