google/xls #614: Consider a new "direct" channel type

meheff, 2022-05-05:
Currently we have two kinds of channels:
* streaming : FIFO semantics
* single-value : behaves as a register where it holds the last written value and reads are non-destructive.

It may make sense to add a third channel type called "direct" (open for better name suggestions). A direct channel is external only (kSendReceive not supported) and lowers to a direct connection to a top-level port. A direct kReceiveOnly channel can have multiple receives (this results in fan-out from the corresponding input port). While a direct kSendOnly channel can have only a single send instruction.

The motivation for this is to support connections from the outside into arbitrarily parts of the design. A common use case might be configuration bits which are set once and do not change during the execution. In general, changing the values on direct channels during execution will have undefined behavior because it will be asynchronous to proc ticks.

This might be used to simplify the IR generated by xlscc. xlscc currently uses single value channel for this purpose, but the register semantics of single-value channel is not exactly what is desired. Rather the intention is the that channel will lower to a wire. However, analysis to prove that the single-value channel can safely lower to a wire is non trivial (and must be performed by proc inlining). Perhaps xlscc can use this new direct channel construct to more clearly communicate the intention.

With this new channel type it may make sense to restrict single-value channels to kSendReceive only (only between procs) as the difference between a single-value external channel and a direct external channel is not clear. So we'd have the following channel types.

* *streaming*: internal or external. Behaves as an arbitrary depth FIFO. Lowers to data port with flow control for external channels, and a fifo with flow control at both ends for internal channels.

* *single value*: internal only (kSendReceive). Behaves like a register and lowers to one.

* *direct*: external only. Indicates a direct connection to a top level port.

meheff, 2022-05-13:
Here's a motivating example from xlscc. The example uses single-value channels to shuttle information to and from other procs under the assumption that they will lower to wires (not saved on state). However, as mentioned above analysis proving the safety of lowering of single-value channels to wires is tricky. This example walks through changing these single-value channels to streaming channels. The hope is that using streaming channels won't result in bloated state.

In this case we have a proc A which sends data to proc B which iterates some number of times then sends data back. This might be generated by a function containing a loop. The function becomes proc A and the loop becomes proc B. Currently xlscc might generate something like:

Channels:
  `stream_in` : external input streaming channel (data is not too big)
  `stream_out` : external output streaming channel (data is not too big)
  `singlevalue_in` : external input single value channel (data is large and invariant)
  `a_to_b_sync`/`b_to_a_sync` : internal streaming channel, empty tuple type
  `a_to_b_data`/`b_to_a_data` : internal single-value channel (data is large as it contains `singlevalue_in` channel data)

Proc A:

```
st = {...}
while (true):
  x = rcv(stream_in)
  y = rcv(singlevalue_in)
  tmp1 = f(st, x, y)
  snd(a_to_b_sync, ())
  snd(a_to_b_data, tmp1)

  tmp2 = rcv(b_to_a_data)
  rcv(b_to_a_sync)
  tmp3 = g(st, tmp2, x, y)
  send(stream_out, tmp3)

  st = h(st, tmp2, x, y)
```

Proc B:

```
count = 0
st = {...}
while (true):
   rcv_if(a_to_b_sync, count == 0)
   data = rcv(a_to_b_data)
   tmp = f'(data, st)
   snd(a_to_b_data, tmp)
   snd_if(a_to_b_sync, (), count == MAX)

   count = count == MAX ? 0 : count + 1
   st = g'(st, data, count)
```

In this case, after proc inlining `x` will like be saved to state because `snd(a_to_b_data)`, a send over a single-value channel, is (presumably) dependent upon `x` via the computation `f(...)`. Also, `send(stream_out)` being dependent upon `x` will similarly result in `x` saved to state. Note that the large data coming from `single_value_in` and subseqently sent on `a_to_b_data` and `b_to_a_data` is never saved to state.

Consider changing xlscc to instead send data via streaming channels to the loop proc B. The sync channels are no longer needed:

Proc A:

```
st = {...}
while (true):
  x = rcv(stream_in)
  y = rcv(singlevalue_in)
  tmp1 = f(st, x, y)
  snd(a_to_b, tmp1)

  tmp2 = rcv(b_to_a)
  tmp3 = g(st, tmp2, x, y)
  send(stream_out, tmp3)

  st = h(st, tmp2, x, y)
```

Proc B:

```
count = 0
st = {...}
saved_data = {...}
while (true)
   data = rcv_if(a_to_b, count == 0)
   saved_data = count == 0 ? data : saved_data
   tmp = f'(saved_data, st)
   snd_if(a_to_b, tmp, count == MAX)

   count = count == MAX ? 0 : count + 1
   st = g'(st, saved_data, count)
```

In this variant, the proc B as constructed explicitly saves the data sent on the `a_to_b` channel. The makes it available for all subsequent iterations of proc B. The potential downside is that this data is *large* by assumption. State optimizations would *have* to remove unnecessary state values to avoid bloat in flop count. This would have to occur after proc inlining because `saved_data` (or some value derived from it) is sent back to proc A so it appears all of this data is used  when the scope of the analysis is restricted to proc B.  Ater proc inlining the scope of the analysis would cover both proc A and B's logic. Assuming `saved_data` is a compound type (eg. tuple), you can decompose it into its constituent leaves and have something like:

Proc with A and B inlined into it:

```
B_saved_data_0 = ...
B_saved_data_1 = ...
B_saved_data_2 = ...
B_saved_data_3 = ...
...
while (true)
  ...
  a_to_b_data = ... // inlined channel a_to_b
  B_saved_data_0 = count == 0 ? a_to_b_data[0] : B_saved_data_0
  B_saved_data_1 = count == 0 ? a_to_b_data[1] : B_saved_data_1
  B_saved_data_2 = count == 0 ? a_to_b_data[2] : B_saved_data_2
  B_saved_data_3 = count == 0 ? a_to_b_data[3] : B_saved_data_3
  ...

  // only a subset of the B_saved_data elements are used.
  ... = .. B_saved_data_0 ...
  ... = .. B_saved_data_2 ...
```    

In this case it should be easy to identify the unused state elements as their only side-effecting uses are in the computation of their own next value. These unused elements can then be removed from the proc state.

Another potential possibility is that the state element is used, but comes from an invariant source (such as an external single-value channel). In this case, saving the value to state is unnecessary because it never changes. Removing such a state element is a bit trickier though because you have to prove that the initial value is not visible. Proving visibilty can be tricky and it's unclear how hard it might be in this case.

So likely we can remove most of the unnecessary state (e.g, unused state elements) and potentially all of it (requires visibility analysis) for this case. So changing xlsccc to using streaming channels is likely something we want to do.

spurserh, 2022-06-06:
This commit removed the use of single value channels for proc to proc communication, using only streaming ones https://github.com/google/xls/commit/abc1c4438d56c588a814bb180d481e005cc8d57c

closed 2022-06-06 by spurserh