google/xls #617: [DSLX] Support methods on structs

cdleary, 2022-05-10:
As reflected in #526 an outstanding request has been to support methods on struct instances so folks can use the familiar "object oriented receiver" style function invocations -- without polymorphism the additional utility is limited but familiarity/convenience / some conciseness will certainly be increased, and it helps pave the path towards monomorphization, right now the DSL only supports parametric instantiation via constexpr values.

taktoa, 2023-01-27:
+1, this would be really nice. I was feeling this pain recently.

proppy, 2024-09-30:
related https://github.com/google/xls/commit/1af3c8cf8144f46be9bca6e44b98477adfe5dd8d

allight, 2026-11-14:
Closing this issue as fixed (accepted by @erinzmoore): DSLX now supports Rust-style `impl` blocks on structs (`Impl` in `xls/dslx/frontend/ast.h` and `ParseImpl` in `xls/dslx/frontend/parser.cc`) with methods (`self`) and associated constants, tested in `xls/dslx/tests/impl.x` and `xls/dslx/tests/parametric_impl.x`. (Closed by Gemini)

closed 2026-11-14 by allight