google/xls #626: [Optimizer]Rewrite rule for multiplexer to reduce total area

zihaoli-cn, 2022-05-22:
Motivating Example:
- `t = Sel(cond, Add(x1, x2), Add(y1, y2), Add(z1, z2))` is equivalent to `t = Add(Sel(cond, x1, y1, z1), Sel(cond, x2, y2, z2))`.

The delay along this `t`'s path is the same. But the latter one will require less area. Some experienced Verilog programmers would do such transformation by hand when they find a chance.

To formally describe it, if the cases of a Select-Type node's operands are homogenous, i.e., they have the same `op` and `type`, they can be rewritten according to the former example.

But it is not always beneficial to do such a transformation. To make sure it is beneficial, it may need to use the `AreaEstimator` to build up a cost model.

taktoa, 2022-05-23:
The `mutual_exclusion_pass` I wrote is designed to do this. Once extended to support merging more operations (I have pending work that does this), it will partition the set of all similar nodes that don't have transitive dependencies between them into the fewest number of sets that contain nodes that are all mutually exclusive (via graph coloring on the complement graph). This would result in

```
fn bar(x_1: bits[32], x_2: bits[32], y_1: bits[32], y_2: bits[32], cond: bits[1]) -> bits[32] {
sel_op_1: bits[32] = sel(cond, cases=[x_1, y_1], id=14)
sel_op_2: bits[32] = sel(cond, cases=[x_2, y_2], id=15)
add: bits[32] = add(sel_op_1, sel_op_2, id=16)
ret useless_sel: bits[32] = sel(cond, cases=[add, add])
}
```

which can then further be optimized by removing `useless_sel`.

The current `mutual_exclusion_pass` only works on send and receive nodes as this has no area impact and is necessary for some of our benchmarks to go through codegen. Area estimation will be an important part of the story to extend this to nodes that have area.

taktoa, 2022-05-23:
See also https://github.com/google/xls/issues/552

zihaoli-cn, 2022-05-23:
Cool!

taktoa, 2022-05-23:
Closing since this is already covered by #552

closed 2022-05-23 by taktoa