google/xls #634: Support non-zero start indexes in DSLX for loop

vincent-mirian-google, 2022-06-06:
The following code is a simplified version of the use case and is not the best representation of the use case.
```
fn main() -> u32[8] {
  let osl_entry: u32[8] = u32[8]:[0,0,0,0,0,0,0,0];
  let osl_entry = for (i, array):
    (u32, u32[8])
    in range(u32:1, u32:8) {
      update(array, i, i)
  }(osl_entry);
  osl_entry
}
```

When executed with xls/dslx/interpreter_main, gives:
```
ConversionError: For-loop is of an unsupported form for IR conversion; only a range(0, CONSTANT) call is supported; got iterable: range(u32:1, u32:8)
```

RobSpringer, 2022-06-13:
I'm shocked that this is still the case. I should be able to handle this one without too much trouble (...I think. Famous last words and all.).

RobSpringer, 2022-06-13:
Oh...unless this is an IR-level restriction. I'll look into it.

closed 2022-06-27 by RobSpringer