google/xls #645: Add partial product multiply IR op

meheff, 2022-06-22:
Currrently XLS IR has two multiply operations, `kUMul` and `kSMul`, for unsigned and signed multiplication. They support arbitrary, mixed widths for the two operands and the result. It would be useful to have partial product multiply operations as well.  These operations would perform only part of the multiply operation leaving the final summation to be performed by a seperate add operation. A motivation is that this enables splitting multiplies across cycles enabling more flexibility in meeting timing. Efficient hardware implementations of partial product multiplies are provided by third party vendors and XLS could potentially lower to an instantiation of such an implementation.

The ops might look like:

```
result = umulp(lhs, rhs)
result = smulp(lhs, rhs)
```

Where the types are:

Value    | Type
-------- | ---------
`lhs`   | `bits[M]`
`rhs`   | `bits[N]`
`result`   | `(bits[R], bits[R])`

The bitwidths `M`, `N`, and `R` can be arbitrary. The operation produces a tuple containing two partial products where the sum of the partial products is the (un)signed product of `lhs` and `rhs` _though the specific values of each element are not defined_ (implementation-defined behavior). This matches the behavior of third party IP which may choose different implementations depending on width and technology and the different implementations may produce different partial product values.

Potential complications:

1. Implementation-defined semantics are new to XLS and introduce the possibilty of skew between different levels of evaluation (DSLX, IR, and Verilog). This could be mitigated by raising an error in the verifier if the result of this operation is used anywhere outside of an add of the two elements (after potentially flowing through flops) because the sum of the two elements do have consistent semantics.

1. The JIT or interpreter might produce a tuple containing zero and the product of the lhs and rhs.  For the purposes of fuzzing it might be useful for these implementations to emit partial products which vary randomly (though still satisfy the property that their sum is the product of the operands).

1. Codegen needs to support instantiating third party Verilog modules (or macros?) for the implementation of the partial product op in Verilog. One possibility is to follow the example of the gate IR op (and a couple others) where a user specifies a format string which is filled in at codegen time:
https://github.com/google/xls/blob/b653c3d054adb015146f4489895f58cc5a129f8d/xls/codegen/codegen_options.h#L175
Some potential quirks are that the instantiated module has multiple outputs so the template string would have to declare a couple temporaries, pass them to the instantiation, then concatenate them together in an assignment to create the output of the op.

1. `umulp/smulp` could be optimized similar to normal multiplies in some cases. Constant folding and multiply-by-one or multiply-by-zero could be replace `umulp/smulp` with one of their operands or a constant. Also narrowing could reduce the width of the operations.

1. After scheduling (in a block) if a `umulp/smulp` is followed by an add the operation could be replaced with a normal multiply.

1. In the future, the splitting of a regular multiply into its partial-product form plus an adder could be performed automatically when necessary as a scheduling optimization.

1. For delay estimation, initially the partial product multiplies could just have the same delay as a regular multiply. Or better yet, the partial product forms could be characterized separately or defined as the delay of a normal multiply less the delay of an R-bit adder.

1. At the DSLX level we probably want to implement this as a builtin.

1. LEC could be broken by these new ops due to their implementation-defined semantics. However, large multipliers which this op is targeting often don't LEC in reasonable time so perhaps this is not that big of a loss.

This could be the first of several future operations which use a [redundent representation](https://en.wikipedia.org/wiki/Redundant_binary_representation). Another usecase might be carry save adders.
 Or perhaps (?) when XLS has Verilog FFI is supported in XLS this operation will no longer necessary because users can instantiation partial product modules directly.

grebe, 2022-07-13:
I had a version that:
- Added the OP to the relevant places
- Codegen'd something similar to `umul` and `smul` with a function that returned `(0, x*y)`
- If a format string was set in `CodegenOptions`, it would replace the body of the function with a format string.

However, there was a big problem with this approach that I only realized upon testing. You can't instantiate a module in a function body, so the primary use-case of using an IP to implement the partial product wouldn't work. The existing things that codegen custom user format strings don't seem to instantiate modules- asserts have their own section with assert statements, and gates declare a variable and assign to it. Neither of those are a clean fit for the module instantiations that partial product multiplies need.

I'm currently working on generalizing how codegen deals with overriding op codegen, and will create flavors for the existing use-cases as well as for overriding op codegen by instantiating a module.

grebe, 2022-07-29:
One potential issue is that some split multiply implementations may have weird constraints on the output width, e.g. `output_width > lhs_width + rhs_width`. It seems bad for narrowing passes to have to respect this, but it's also not clear how to deal w/ this in codegen or via format strings.

grebe, 2023-05-03:
This is better solved in a wrapper than by avoiding narrowing.

closed 2023-05-03 by grebe