google/xls #649: Build of xlscc fails

joachimschmidt557, 2022-07-01:
With the latest version of the main branch (a93c40be378b1942657c02c5689d38fb78ff0f5f at the time of writing), building `//xls/contrib/xlscc/...` fails with the following message:

```
ERROR: /home/joachim/xls/xls/contrib/xlscc/BUILD:120:11: Compiling xls/contrib/xlscc/translator.cc failed: (Exit 1): gcc failed: error executing command /usr/bin/gcc -U_FORTIFY_SOURCE -fstack-protector -Wall -Wunused-but-set-parameter -Wno-free-nonheap-object -fno-omit-frame-pointer '-std=c++17' -MD -MF ... (remaining 146 arguments skipped)

Use --sandbox_debug to see verbose messages from the sandbox and retain the sandbox build root for debugging
xls/contrib/xlscc/translator.cc: In member function 'absl::StatusOr<xlscc::CValue> xlscc::Translator::GenerateIR_Call(const clang::CallExpr*, const xls::SourceInfo&)':
xls/contrib/xlscc/translator.cc:2544:8: warning: variable 'sequencing_safe' set but not used [-Wunused-but-set-variable]
 2544 |   bool sequencing_safe = false;
      |        ^~~~~~~~~~~~~~~
xls/contrib/xlscc/translator.cc: In member function 'absl::StatusOr<xlscc::CValue> xlscc::Translator::GenerateIR_Expr(const clang::Expr*, const xls::SourceInfo&)':
xls/contrib/xlscc/translator.cc:3447:34: error: 'const class clang::StringLiteral' has no member named 'isOrdinary'
 3447 |       if (!(string_literal_expr->isOrdinary() ||
      |                                  ^~~~~~~~~~
In file included from external/com_google_absl/absl/base/macros.h:36,
                 from external/com_google_absl/absl/algorithm/container.h:54,
                 from external/com_google_absl/absl/container/flat_hash_map.h:38,
                 from ./xls/contrib/xlscc/translator.h:24,
                 from xls/contrib/xlscc/translator.cc:15:
./xls/common/logging/check_ops.h: In instantiation of 'std::string* xls::logging_internal::Check_EQImpl(const T1&, const T2&, const char*) [with T1 = char [29]; T2 = std::nullptr_t; std::string = std::__cxx11::basic_string<char>]':
xls/contrib/xlscc/translator.cc:2859:9:   required from here
./xls/common/logging/check_ops.h:200:46: warning: the compiler can assume that the address of 'v1' will never be NULL [-Waddress]
  200 | XLS_LOGGING_INTERNAL_CHECK_OP_IMPL(Check_EQ, ==)
external/com_google_absl/absl/base/optimization.h:178:58: note: in definition of macro 'ABSL_PREDICT_TRUE'
  178 | #define ABSL_PREDICT_TRUE(x) (__builtin_expect(false || (x), true))
      |                                                          ^
./xls/common/logging/check_ops.h:200:1: note: in expansion of macro 'XLS_LOGGING_INTERNAL_CHECK_OP_IMPL'
  200 | XLS_LOGGING_INTERNAL_CHECK_OP_IMPL(Check_EQ, ==)
      | ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
```

Reverting a42bd1344a4b5aa9cfda9442207c6f81c43a7617 fixed this issue. `xlscc` is skipped in the CI, so this error wasn't caught there.

RobSpringer, 2022-07-01:
Hey Joachim!

Thanks for the report. I *thought* I fixed this yesterday, but apparently
not. I just checked in rev 4dbb13f
<https://github.com/google/xls/commit/4dbb13f71d9993f9c4087a7dda005802f21c5f53>,
which should definitely for real this time resolve the issue. Let me know
if you still see breakage!
-rob

joachimschmidt557, 2022-07-03:
Hi Rob!

As of 4dbb13f71d9993f9c4087a7dda005802f21c5f53, building `xlscc` works again. Thanks!

closed 2022-07-03 by joachimschmidt557