google/xls #651: trace_fmt should support printing bitfields with leading zeros

sandwichmaker, 2022-07-07:
Currently if you have a variable `x` that is a `u8` and has value 3, printing it using `trace_fmt!("{:b}",x) `will print out `11`. 
It would be nice to be able to say  `trace_fmt!("{:0b}",x)` and get `00000011` instead. 

This is particularly useful when debugging fixed point arithmetic.

hzeller, 2023-10-24:
+1, I just ran into a need for that as well.

dplassgit, 2024-07-24:
> Potential follow on features:
> 
> `{:0x}` : zero-padded hex numbers

This is already in progress.

> (I don't think we need zero padded decimal)

Zero padded decimal is expressly forbidden: https://github.com/google/xls/blob/3d9d002b1ca12fc7e6032202c08a858862b05266/xls/ir/bits_ops.cc#L665

commit 61bb6d4e (2024-07-29): Add `{0x}` format for `trace_fmt!` which can format hex numbers with leading zeros.

https://github.com/google/xls/issues/651

Fixes #651

PiperOrigin-RevId: 657262247
closed 2024-07-29 by copybara-service[bot]