google/xls #658: Workflow to analyze and optimize individual stages

grebe, 2022-08-09:
Today @meheffernan and I talked about ways to optimize the critical path and number of stages. One capability that seemed useful was extracting a pipeline stage, running it through an implementation flow, and iterating on the contents of the stage.

One useful way to use this capability is to greedily fill stages until they no longer meet timing in an attempt to find the minimum number of stages that meet timing. This isn't perfect as it ignores fanout effects, but is useful to answer questions about how many stages are required to meet timing.

Another useful way to use this capability is to take a scheduled design, extract a stage that is failing timing, and prune things from the stage until it meets timing. This could be fed back to the scheduler (various ways to do this seem reasonable, e.g. a delay model that annotates delays on certain nodes and falls back on another delay model if unannotated, or timing constraints that vary per-stage to account for delay model error, or "fence' ops). The process of iteratively pruning can also be used to find places where the delay model needs to be improved.

cdleary, 2022-10-28:
It this saying the greedy-adding version another way? Imagine we add single XLS ops at a time to a pipe stage (in N^2 kind of fashion) and after each addition ask the underlying RTL synthesis tool (e.g. one that did logic replication for timing closure and other interestingly non-linear things) whether it still closes timing. We could converge on the maximally packed pipe stage possible by doing that and use that to help cross-validate delay modeling. Am I understanding the idea right? I always thought something like this would be a great way to refine the delay model understanding quickly according to the dominant real considerations taken into account by the underlying RTL synthesis tool.

grebe, 2022-11-03:
Yes, what you're describing is what I was trying to say.

There are still some issues not fully addressed by this idea, like you probably don't not want to live on the one extreme of the area/delay curve and clock relaxation is a pretty coarse tool to achieve this. I think the same underlying tools (iterate on a packing of a pipeline stage, run through the whole backend flow and extract PPA, etc.) can also be used to explore more general PPA tradeoffs.