google/xls #661: Enable send/recv on Values in JitChannelQueue

RobSpringer, 2022-08-16:
Currently, the send and receive calls in JitChannelQueue operate on raw buffers + len, e.g. `void JitChannelQueue::Send(const uint8_t* data, int64_t num_bytes);`. It might make users' lives easier if they could send/receive Values instead or as well.

For internal proc communication, raw buffers are as useful as needed, but for users inserting values into and getting values out of queues (think proc network inputs & outputs), this is a pain, as they need to be aware of endianness, etc.

hongted, 2022-08-16:
Currently we have the JitChannelQueueWrapper to help with this.  Only reason it wasn't incorporated into JitChannelQueueWrapper is the need to have access to the ProcJit object. 
 
https://github.com/google/xls/blob/main/xls/flows/ir_wrapper.h#L38

meheff, 2023-06-08:
Channel queues in the jit and interpreter now all derive from the same base class and support this feature.

closed 2023-06-08 by meheff