google/xls #664: Clarify valid channel uses/semantics

RobSpringer, 2022-08-17:
Similar to issue #662 : can I pass a channel to a function? Can I recv on it inside a function? What other potential restrictions are there on channel usage?

vincent-mirian-google, 2022-08-22:
Also, what does it mean as a member of a tuple?

```
proc main {
  x1: chan in u2;
  x5: chan in s8;
  x11: chan in s15;
  config(x15: chan in u2, x16: chan in s8, x17: chan in s15) {
    (x1, x5, x11)
  }
  next(x0: token) {
    let x2: (token, u2) = recv(x0, x1);
    let x3: token = x2.0;
    let x4: u2 = x2.1;
    let x6: (token, s8) = recv(x3, x5);
    let x7: token = x6.0;
    let x8: s8 = x6.1;
    let x9: (s8, token, s8, chan in s8, token, (token, s8)) = (x8, x0, x8, x5, x0, x6); // HERE
    let x10: bool = (x4) > (x4);
    let x12: (token, s15) = recv(x3, x11);
    let x13: token = x12.0;
    let x14: s15 = x12.1;
    ()
  }
}
```

RobSpringer, 2022-09-02:
From my recent work, it sure looks like passing channels to functions would be helpful - otherwise, you've got to place a lot of logic in your proc's `next()`, and that tends to get overwhelming quickly.