google/xls #666: DSLX interpreter restricts function unit tests when proc with state is present

RobSpringer, 2022-08-17:
If I have a function and a proc (that has a state parameter) defined in the same file, and a unit test for that function but no unit test for the proc, the interpreter will not execute the function unit test. The interpreter searches for a "top proc" in any file (for creating boundary channels), and if one is found, it tries to bind an initial value to that proc. If one isn't specified, then the interpreter will exit, citing the lack of an initial value for that top-level proc.

I don't have an immediate solution to this; perhaps make --top required for IR conversion of procs?

vincent-mirian-google, 2022-08-18:
Is there a manner to provide a initial state to the proc in the interpreter? I believe it is posible from the IR converter.

If the proc is not used in the interpreter, we should probably not impose an initial state. That's my thoughts.

I am open to understand the issue further and implement a solution.

allight, 2026-11-14:
Closing this issue as fixed (accepted by @allight): Mandatory in-source proc `init` blocks (`#681`) and per-entity test execution in `xls/dslx/run_routines/run_routines.cc:1027-1065` eliminated the `--top_proc_initial_state` requirement that previously blocked running function unit tests when a stateful proc was present in the same DSLX file. (Closed by Gemini)

closed 2026-11-14 by allight