google/xls #670: Compiling failed: 'No module named google.protobuf'

amgawishx, 2022-08-22:
I followed the tutorial compiling steps to the letter, yet I keep getting the same error:
[image]
Despite both google and protobuf being installed on python and the module `google.protobuf` imports normally if I try it myself.
So, what could be causing this?
System info:
Ubuntu 22.04 x86_64
5.15.0-46-Generic
Oracle VirtualBox VM

cdleary, 2022-08-22:
Thanks for reporting, I'll try to repro. In the meantime the docker-based installation should be able to get you going in a known-working way.

cdleary, 2022-08-23:
I'm sorry but I was unable to reproduce this:

[image]

I installed a fresh Ubuntu 22.04 from ISO in virtualbox 6.1, updated, installed bazel via the website instructions linked in the XLS README, ran the apt commands listed in the bazel install website, apt installed git, cloned, and bazel tested everything -- all successful.

Trying to think of how my procedure may have been different -- is your bazel version 5.2? Did you run the apt commands from the README?

(Note that, inside, I believe Bazel maintains a virtual Python environment where packages get installed, e.g. we have a requirements.txt as part of our Bazel workspace, so what you see in a Python shell _outside_ of the Bazel build environment will not be reflective of what's available inside, you're using your system wide Python site-packages.)

amgawishx, 2022-08-23:
I used the minimalist install of the distro not the full, and here is the script I use:
```sh
#!/bin/bash
apt update 
apt install build-essential libboost-system-dev libboost-thread-dev libboost-program-options-dev libboost-test-dev build-essential python3 python3-venv python3-pip g++ unzip zip apt-transport-https curl gnupg swig cmake docker docker.io git freeglut3-dev libglib2.0-0 ffmpeg
pip install tensorflow-cpu numpy matplotlib dask virtualenv pydot graphviz hls4ml pyspice google Pillow opencv-python ffio shapely descartes pyrender torch torchvision pickle5 h5py vista xls
curl -fsSL https://bazel.build/bazel-release.pub.gpg | gpg --dearmor >bazel-archive-keyring.gpg
mv bazel-archive-keyring.gpg /usr/share/keyrings
echo "deb [arch=amd64 signed-by=/usr/share/keyrings/bazel-archive-keyring.gpg] https://storage.googleapis.com/bazel-apt stable jdk1.8" | tee /etc/apt/sources.list.d/bazel.list
apt update
apt install bazel
apt install python3-distutils python3-dev libtinfo5 python-is-python3 libncurses5
git clone https://github.com/google/xls.git xls
cd xls
brazel build -c opt //...
```

cdleary, 2022-08-23:
Can you point me at the ISO download page? I thought the server-class Ubuntu installation didn't have a window manager by default but you had one in your screenshot so I just want to make sure I'm getting it right.

amgawishx, 2022-08-23:
https://ubuntu.com/download/desktop/thank-you?version=22.04.1&architecture=amd64
I used this, it was a desktop version minimalist installation

cdleary, 2022-08-25:
This was minimal installation and it worked as well:

[image]

```
cdleary@virtualbox:~/xls$ history 
    1  cd /media/cdleary/VBox_GAs_6.1.36/
    3  ./autorun.sh 
    4  sudo apt install apt-transport-https curl gnupg
    5  curl -fsSL https://bazel.build/bazel-release.pub.gpg | gpg --dearmor >bazel-archive-keyring.gpg
    6  sudo mv bazel-archive-keyring.gpg /usr/share/keyrings
    7  echo "deb [arch=amd64 signed-by=/usr/share/keyrings/bazel-archive-keyring.gpg] https://storage.googleapis.com/bazel-apt stable jdk1.8" | sudo tee /etc/apt/sources.list.d/bazel.list
    8  sudo apt update && sudo apt install bazel
    9  sudo apt-get install git
   10  git clone https://github.com/google/xls.git
   11  cd xls/
   12  sudo apt install python3-distutils python3-dev libtinfo5 python-is-python3
   14  bazel test -c opt //...
   16  cat /var/log/installer/telemetry | egrep -i Minimal
```

I'll try to install that unrelated stuff now that you have in your commands to see if it interferes, I guess that'd have to be the explanation, but given bazel makes a virtualenv inside it'd be quite confusing.

cdleary, 2022-08-25:
Something in the pip line tries to install PyV8 and there's no header installed for it:

[image]

That being said, this is getting to a level of build/environmental debugging we don't generally do in the core development of the project, so I think I'm going to mark as cannot repro -- I can suggest you can build the XLS binaries you need inside a Docker container and they should be fully linked such that they work in your external environment without a need for dependency support. I hope that makes a workable path for you!

closed 2022-08-25 by cdleary

tcal-x, 2023-02-15:
@amgawishx , I recently had a similar issue.   I realized I had done a local `pip install` not in a virtual env at some point in the past and had files at `/usr/local/lib/python3.10/dist-packages/google`.   I uninstalled these with `sudo pip uninstall google`, and that resolved the issue for me.