google/xls #673: DSLX: IR converter attempts to convert test procs?

RobSpringer, 2022-08-22:
Get the following error when running the GCM GHASH test (in flight; will be submitted soon) DSLX interpreter *without the --compare=none* flag:

```
xls/modules/aes/aes_128_ghash.x:334:26-334:41                                                                     
0332:             [u8:0x9e, u8:0x5e, u8:0x57, u8:0x14],                                                                       
0333:         ];                                                                                                              
0334:         let _ = assert_eq(tag, expected);                                                                               
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^-------------^ ConversionError: Could not find name for invocation: assert_eq; available: [block_to_u128, u128_to_block, gf128_mul, xor_block, get_current_state, aes_128_ghash.config, aes_128_ghash.next, aes_128_ghash_test.config, aes_128_ghash_test.next]                                                                                           
0335:                                                                                                                         
0336:         // Test 2: two AAD blocks, three ctxt blocks. Random data. 
```

Looks like it's trying to IR-convert the test?

proppy, 2023-05-16:
/cc @mtdudek

proppy, 2023-05-16:
That doesn't seems to be limited to procs, the same error would happen when trying to convert a regular non-`[test]` annotated `fn` that calls `assert_eq`:
```
fn f1() -> u1 {
    u1:1
}

fn testf1() {
    let _ = assert_eq(f1(), u1:1);
    ()
}
```
would returns:
```
F0516 14:25:21.382737  420816 ir_converter_main.cc:163] Check failed: ::absl::OkStatus() == (status) (OK vs. INVALID_ARGUMENT: ConversionError: f1.x:7:22-7:34 Could not find name for invocation: assert_eq; available: [f1, testf1]) 
```

@cdleary is there something special done to exclude `[test]` from conversion?

proppy, 2023-05-16:
We did note a different between `#[test]` and `#[test_proc]` with `--v=6`:

`#[test_proc]` do print:
```
I0516 15:15:48.596359  432009 extract_conversion_order.cc:786] Adding to ready sequence: main.next
```
while `#[test]` function never show this message.

Looking deeper in the test collection logic:
https://github.com/google/xls/blob/7901203b7b48d357f5861ca6e293111498c9935b/xls/dslx/ir_convert/extract_conversion_order.cc#L886-L899
It seems that it manipulate the `ready` vector directly rather than calling `AddToReady` / `IsReady` methods, the later do have some early return condition when handling `TestFunction` which don't seem to have an equivalent for `TestProc`:
https://github.com/google/xls/blob/7901203b7b48d357f5861ca6e293111498c9935b/xls/dslx/ir_convert/extract_conversion_order.cc#L886-L899
 https://github.com/google/xls/blob/7901203b7b48d357f5861ca6e293111498c9935b/xls/dslx/ir_convert/extract_conversion_order.cc#L702-L706

proppy, 2023-05-16:
From a quick pass thru the code it appeared that `TestFunction` would never be collected by `traverse_tests` because of the early return in `IsReady`.

allight, 2026-11-14:
Closing this issue as fixed (accepted by @allight): `ConvertOptions::convert_tests` (`xls/dslx/ir_convert/convert_options.h`) and `GetOrderForEntry` (`xls/dslx/ir_convert/extract_conversion_order.cc`) now skip `#[test]` and `#[test_proc]` constructs during non-test IR conversion. (Closed by Gemini)

closed 2026-11-14 by allight