google/xls #681: DSLX: Remove proc initial value specification from the BUILD flow

RobSpringer, 2022-08-29:
To convert a DSLX proc into IR, we need to specify an initial value for the proc's state. That specification is currently done in the BUILD file, which is fragile. Consider the following example:
```
xls_dslx_opt_ir(
    name = "aes_128_ctr",
    dslx_top = "aes_128_ctr",
    ir_conv_args = {
        "top_proc_initial_state": "'(0, (0, [0, 0, 0, 0], 0, 0), 0, 0)'",
    },
    library = ":aes_128_ctr_dslx",
)
```

First of all, this is simply hard to read or parse for human eyes, and secondly, it's hard to maintain: what if the state changes for aes_128_ctr? We need to figure out some way to remove initial values from this flow.

cdleary, 2022-08-29:
I'd think we need a construct on the top level proc in the DSL and it could just be extracted as part of IR conversion? Is there a natural place to put that?

vincent-mirian-google, 2022-10-06:
From #735 

A thought is to have the initial values declared in a 'parameter list' for the proc.

In the following example, 'main(**u32: 1**)' has the inital values.

```
proc main(u32: 1) {
  c: chan<u32> out;

  config(c: chan<u32> out) {
    (c)
  }

  next(tok: token,  a: u32) {
    let tok = send(tok, c,  a);
    let a = a + u32:1;
    (a)
  }
}
```

RobSpringer, 2022-10-20:
Part of this: enable proc-scoping of constants, such that we can define a default (but overrideable) initial state for the proc that doesn't leak out (e.g. if we define multiple procs in the same file, having to proc-name-prefix their default initial states seems clumsy).

RobSpringer, 2022-10-24:
I've a patch out (internally) to enable specifying constants as initial values, but this will need to be expanded to support ColonRef'ed constants.

allight, 2026-11-14:
Closing this issue as fixed (accepted by @allight): DSLX procs now define their initial state directly in source via `init { ... }` (or `ProcDef` `new`), and `top_proc_initial_state` was removed from `xls/build_rules/xls_ir_rules.bzl` and `xls/dslx/ir_convert/ir_converter.cc`. (Closed by Gemini)

closed 2026-11-14 by allight