google/xls #691: DSLX: No way to do for-loop parallel sends

RobSpringer, 2022-09-02:
Say I have N components that I want to send to in parallel. I'll need a way of collecting their tokens so we can join on them later.

Something like:
```
  const NUM_THINGS = u32:4;
  let toks = token[NUM_THINGS]:[???, ...];
  for toks = for (i, toks) in range(0, NUM_THINGS) {
      let x = send(tok, chan[i], foo);
      update(toks, i, x);
  }(toks);
  let tok = join(toks);
```

1. There's no _clean_ way to declare a literal token. You can fudge it by doing `join()` with no arguments.
2. You can't declare literal token arrays: `TypeInferenceError: uN[4] Annotated type for array literal must be an array type; got ubits token[4]`.
3. `join()` doesn't accept arrays.

hongted, 2023-05-08:
Operations that generate implicit tokens (ex. trace_fmt, cover, etc...) also need the ability to be placed inside for-loops.

allight, 2026-11-14:
Closing this issue as fixed (accepted by @allight): `xls/dslx/frontend/builtin_stubs.x:70-71, 105` exposes `token()` (`fn token() -> token;`) and variadic `join(...)` (lowered in `xls/dslx/ir_convert/function_converter.cc:3034`), and `xls/dslx/type_system_v2/type_annotation_utils.cc:306-308` supports `token[N]` array type annotations, allowing tokens to be created and joined across `for`-loop iterations for parallel sends. (Closed by Gemini)

closed 2026-11-14 by allight