google/xls #692: DSLX: Bad error on `send` on channel array (instead of a channel)

RobSpringer, 2022-09-02:
I accidentally did:
```
my_chans: chan<u32>[4];
...
let tok = send(tok, my_chans, foo);
```

Obv. wrong: you send on individual channels, not channel arrays (tho that's an interesting idea). Instead of a nice error message, I got a XLS_RET_CHECK failure: `Check failed: ::absl::OkStatus() == (status) (OK vs. INTERNAL: XLS_RET_CHECK failure (xls/dslx/deduce.cc:2192) channel_type != nullptr ((nil) vs. nullptr)`.

Fix plz

xls-interested, 2022-11-04:
Hi Rob,

Is there do you know if there is any support for array of channels in xlscc (and if so how to specify).

Thanks

allight, 2026-11-14:
Closing this issue as fixed (accepted by @allight): `TypeSystemV2` (`xls/dslx/type_system_v2/typecheck_module_v2_proc_test.cc`) and `xls/dslx/type_system/proc_typecheck_test.cc:207-224` (`TEST(TypecheckTest, CantSendOnNonChannel)`) check that the channel argument to `send` has type `chan<Any> out` and emit a clean `TypeMismatch` error instead of crashing on `XLS_RET_CHECK(channel_type != nullptr)`. (Closed by Gemini)

closed 2026-11-14 by allight