google/xls #693: More readable next statement in IR syntax

taktoa, 2022-09-02:
Currently the `next` list in XLS IR is hard to read, because it is a list of values with no reference to the params being updated. If you want to, say, remove a param from a proc that's in the middle of the list, it is very tedious. I think syntax like this would be easier to read and modify with only modest overhead:

```
top proc main(tkn: token, foo: bits[32], bar: bits[32], baz: bits[32], init={1, 2, 3}) {
  send.1: token = send(tkn, foo, channel_id=0)
  next send.1 {
    foo = bar
    bar = baz
    baz = foo
  }
}
```

rather than

```
top proc main(tkn: token, foo: bits[32], bar: bits[32], baz: bits[32], init={1, 2, 3}) {
  send.1: token = send(tkn, foo, channel_id=0)
  next (send.1, bar, baz, foo)
}
```

epsit03, 2024-10-30:
Well, to address it, I propose an updated `map` syntax in the `next` block, where each variable’s state change is clearly mapped. This approach eliminates ambiguity with direct mappings, making additions or removals straightforward and self-contained:

```
top proc main(tkn: token, foo: bits[32], bar: bits[32], baz: bits[32], init={1, 2, 3}) {
  send.1: token = send(tkn, foo, channel_id=0)
  
  next send.1 -> map {
    foo -> bar
    bar -> baz
    baz -> foo
  }
}
```

> Using 'map' block to define the next state of each parameter

I believe it can offer these benefits:

**Scalability**, since it’s easy to add or remove mappings without disrupting the existing ones, which gives us flexibility as project evolves. Another one can be **clarity**, since by explicitly defining each transition, we enhance readability, making it easier for everyone to understand the flow of the code. Moreover, it implies **Modularity** as this approach simplifies complex state changes, which in turn boosts maintainability and can potentially make codebase more manageable.

allight, 2026-11-14:
Closing this issue as fixed / obsolete due to changes in how `next` nodes are emitted (accepted by @allight): XLS IR replaced positional `next (...)` statements with explicit named `next_value(param=<state_elem>, value=<val>, predicate=<pred>)` nodes (`xls/ir/nodes.h`, `xls/ir/ir_parser.cc:910-950`, `xls/ir/proc.h`). (Closed by Gemini)

closed 2026-11-14 by allight