google/xls #701: Unexpected QoR using channels

xls-interested, 2022-09-06:
The included examples create unexpected RTL / QoR - was expecting to see FIFO references in the optimized IR, and the RTL.
Must be doing something very wrong in the examples, or high potential for user error. 
Does something jump out to you?

Notice the two test cases for the channel are in two separate directories. 
To run: run_func0.sh ( might need to modify and point to your local paths )

[channel.zip](https://github.com/google/xls/files/9500892/channel.zip)

taktoa, 2022-09-06:
Without looking at the files: we don't do proper stitching yet; currently we use the "proc inlining" pass to handle situations with multiple procs by turning them into a single proc that simulates all the procs by communicating over proc state (this currently can only handle proc networks that have FIFOs of size 0 or 1).

taktoa, 2022-09-06:
How are you measuring QoR?

cdleary, 2022-09-06:
Can you paste an example of a) what you are seeing in the output that doesn't seem like your expectation and b) what you'd expect to see instead? The more precise on what you were expecting the better we could help guide.

Notably stitching together with FIFOs happens externally to the leaf-module-level generation that XLS does -- it is currently the responsibility of the external stitching tools to instantiate FIFOs appropriate for the platform that the XLS modules are being embedded inside.

We've been talking about how we might offer stitching out of the box in XLS, but it does not exist at the moment as teams we work with have their own (often diverse) stitching solutions / requirements, including their preferred FIFO generation flows.

xls-interested, 2022-09-07:
I think I am expecting that when an channel interface is specified with read and write functions, it will generate the verilog interface with some form of ready/valid signals. I assume that the FIFO can be stitched outside by connecting to those signals..

taktoa, 2022-09-07:
It seems like the issue may be that you're passing in `--top func0` when you want to be passing in `--top func0_proc` (or no `--top` at all).

xls-interested, 2022-09-08:
Thanks. specifying -top func0_proc to both opt and codegen does solve the problem...
However, for the other testcase with for loop (CHANNEL_LOOP), the generated verilog file has port names that do not seem correct.

taktoa, 2022-09-08:
I think you need to add the `--inline_procs` flag.

xls-interested, 2022-09-08:
Thanks. that fixes it.

xls-interested, 2022-09-08:
I have an updated testcase that passes the channel down one function and seems to not like that.. Generate assertion.. Can you help. Testcase attached
[internal_channels.zip](https://github.com/google/xls/files/9530122/internal_channels.zip)

xls-interested, 2022-09-08:
Note that link for external directory should be copied into this from previous example

taktoa, 2022-09-08:
I haven't run the script, but I noticed that this one also doesn't include `--inline_procs` in the call to `opt_main`.

taktoa, 2022-09-08:
oh, this one only has one proc, so it shouldn't need it

taktoa, 2022-09-08:
looks like the issue is that it generated a function named `top`, which is a keyword in XLS IR

xls-interested, 2022-09-08:
thanks this works. now I have a updated testcase with internal channel. Can you please help..
[internal_channels_2.zip](https://github.com/google/xls/files/9531161/internal_channels_2.zip)

taktoa, 2022-09-08:
You'll need to include the output of the failing command in your comment for me to answer your questions. I can't constantly be downloading your reproducers and running them (especially in this case, where it is hard to reproduce your xlscc invocation due to XLS_INCLUDE_DIRS)

xls-interested, 2022-09-09:
Parsing file 'func0.cpp' with clang...
Generating IR...
UNIMPLEMENTED: IO operations should be on parameters
  --> ./func0.cpp:29:31
   |
29 |     ac_channel<int> &InImage, ac_channel<int> &OutImage) {
   |                               ^

F0908 17:00:13.692925 1340572 opt_main.cc:115] Check failed: ::absl::OkStatus() == (xls::tools::RealMain(positional_arguments[0])) (OK vs. INVALID_ARGUMENT: Expected keyword 'package': Expected token, but found EOF.)

taktoa, 2022-09-09:
I noticed in `top_mod` you have `ac_channel<int> InImage` whereas in the rest of the program you accept it by reference, perhaps that could be related? I don't have much experience with the xlscc frontend so I don't actually know what that error message means exactly.

taktoa, 2022-09-13:
Can I close this issue?

taktoa, 2022-09-16:
Closing, feel free to reopen or make a new issue

closed 2022-09-16 by taktoa