google/xls #704: Lower arrays of channels from DSLX into IR

RobSpringer, 2022-09-07:
DSLX currently supports declaring and using arrays of channels, e.g., `let (ps, cs) = chan<u32>[4];`, but these aren't currently lowerable to IR. Sadly, it's not a trivial task to do this. In XLS IR, channels are top-level package constructs, and they're referenced by literal ID in send/recv nodes. The former can be handled during IR conversion, but the latter cannot.

The only workaround would be to emit conditional sends/recvs for every item in the array, and predicate off the undesired operations, but that's horrible and seriously pollutes the IR. Hopefully we can work something out such that send/recv channel IDs can be BValues and not literals. This'll take some discussion, though.

rw1nkler, 2024-01-02:
Are there any plans to provide support for arrays of channels shortly? Would it be possible to flatten the array of channels when converting DSLX to IR, so that each channel in the array gets a separate ID?

proppy, 2024-05-15:
@meheff @grebe will https://google.github.io/xls/design_docs/proc_scoped_channels/ help with this?

Related #869.

proppy, 2024-05-30:
Note currently lowering a proc that accept arrays of channels fails with:
```
Could not extract a textual position from error message: INVALID_ARGUMENT: Expression @ third_party/xls/examples/matmul_4x4/matmul_4x4.x:120:36-120:50 was not constexpr: `activations_in`
```

Related https://github.com/google/xls/issues/983

meheff, 2024-08-02:
I think supporting arrays of channels where the index is constexpr would not be too hard. For `chan<u32>[3]`  is DSLX you just flatten into three separate channels in the IR. If you want to dynamically index into the channels like so:

```
let i = ...;  // not constexpr
send(tok, chans[i], ...);
```

then that is more complicated. But a first stab at this would be supporting the constexpr case.

Generally, there are probably a number of features we can add to make structural replication easier.

grebe, 2024-08-02:
I had a prototype that augmented proc_config_ir_convert's `ProcConfigValue` with arrays/maps of channels and enhanced constexpr evaluation to work for `unroll_for!` in a `config()`. It let some things work, but was pretty fragile and seemed unscalable.

Proc configs are "evaluate everything constexpr" in a way that DLSX constexpr evaluation isn't really meant to be. I've been thinking about leveraging the DSLX bytecode interpreter for proc  IR conversion. The bytecode interpreter's proc config evaluator is currently specialized for test procs, but I don't think it would be too much to generalize. The bytecode interpreter extracts the whole proc network, so I think it should be able to replace all the stuff that extracts the proc conversion order (also pretty messy) in addition to all the channel array accesses.

commit ddfc45a5 (2024-08-21): Actually unroll unroll_for! loops during type checking.

Previously, this was never being done. The practical use of channel arrays currently requires explicit unrolling.

https://github.com/google/xls/issues/704

Fixes #704

Because a loop needs different unrolled renditions per parametric env, we store those with the corresponding env in TypeInfo, rather than trying to produce an edited AST, as old comments incorrectly implied was happening. This means a visitor that wants to use unrolled loops does in fact need trivial handling for UnrollFor, which retrieves the desired unrolled rendition,  [... cut]
closed 2024-08-21 by copybara-service[bot]