google/xls #716: Cannot do multiple writes to a channel

xls-interested, 2022-09-22:
0921 17:49:51.950360  116738 opt_main.cc:115] Check failed: ::absl::OkStatus() == (xls::tools::RealMain(positional_arguments[0])) (OK vs. INTERNAL: Multiple sends associated with the same channel 'OutImage':

e.g.
          pixval = InImage.read();

            OutImage.write(pixval);
	    pixval += 1;
            OutImage.write(pixval);

[simple_channel.zip](https://github.com/google/xls/files/9621272/simple_channel.zip)

taktoa, 2022-09-22:
We don't support this feature. If you send twice on the same channel, that would involve clocking the output at twice the rate of the block. Instead, you should just make a channel containing a struct, or two channels (i.e. double the number of bits in the channel). The only case where you can safely send on the same channel twice, is if the compiler can prove that the two sends are mutually exclusive (this happens on a "best effort" basis), which they are not in the quoted code.

closed 2022-09-22 by taktoa

xls-interested, 2022-09-22:
Thanks. You are correct on that. I am trying to simplify a test case. In the real example, multiple sends are sent out conditionally (in an if then else statement).

 If I use this code

          if (width > height)
                OutImage.write(pixval);
            else {
                // pixval += 1;
                OutImage.write(pixval);
            }

I run into the following error

F0922 09:32:01.827417  141741 codegen_main.cc:376] Check failed: ::absl::OkStatus() == (xls::RealMain(ir_path, absl::GetFlag(FLAGS_output_verilog_path), absl::GetFlag(FLAGS_output_signature_path), absl::GetFlag(FLAGS_output_schedule_path), absl::GetFlag(FLAGS_output_verilog_line_map_path), absl::GetFlag(FLAGS_output_block_ir_path))) (OK vs. UNIMPLEMENTED: Unhandled node for delay estimation in delay model 'asap7': or_reduce.140: bits[1] = or_reduce(heightp_aid2__1: bits[1], id=140)) 

This works ok if I remove the else part or even remove the second write

taktoa, 2022-09-22:
The mutual exclusion pass is successfully merging together those sends, which involves creating an or_reduce node. The asap7 delay model does not currently handle or_reduce nodes. Feel free to create an issue for that.