google/xls #717: xlscc should order sends/receives to avoid deadlock

meheff, 2022-09-22:
It looks like XLSCC does not sufficiently order sends and receives. Generally, a receive should be ordered before a send if there is a data/control dependency from the send to the receive via other procs. The ordering is done by threading a token from the send to the receive.

For example, below is a snippet from the test `TranslatorTest.ForPipelinedBlank`. The send `send.68` which sends on channel 2 should execute before the receive  `receive.70` on channel 3. This is because, in proc `__for_1_proc` the data on channel 2 is used to compute the data sent on channel 3.

This was causing deadlocks in a refactor of the interpreter.  The refactor picked a sort of the nodes and executed them one-by-one in that order. It happend to schedule `receive.70` before `send.68` so it deadlocked. Before the refactor, the interpreter would eagerly try to execute as much as possible even if it got blocked on one node.

```
proc __for_1_proc(tkn: token, st: (bits[1], bits[32], bits[64]), init={(1, 0, 0)}) {
  tuple_index.30: bits[1] = tuple_index(st, index=0, id=30, pos=[(1,12,7)])
  receive.31: (token, (bits[64], bits[32])) = receive(tkn, predicate=tuple_index.30, channel_id=2, id=31, pos=[(1,12,7)])
  tuple_index.33: (bits[64], bits[32]) = tuple_index(receive.31, index=1, id=33)
  tuple_index.40: bits[64] = tuple_index(st, index=2, id=40, pos=[(1,12,7)])
  tuple_index.39: bits[64] = tuple_index(tuple_index.33, index=0, id=39, pos=[(1,12,7)])
  tuple_index.37: bits[32] = tuple_index(st, index=1, id=37, pos=[(1,12,7)])
  tuple_index.36: bits[32] = tuple_index(tuple_index.33, index=1, id=36, pos=[(1,12,7)])
  sel.41: bits[64] = sel(tuple_index.30, cases=[tuple_index.40, tuple_index.39], id=41, pos=[(1,12,7)])
  sel.38: bits[32] = sel(tuple_index.30, cases=[tuple_index.37, tuple_index.36], id=38, pos=[(1,12,7)])
  tuple.42: (bits[64], bits[32]) = tuple(sel.41, sel.38, id=42, pos=[(1,12,7)])
  invoke.43: ((bits[64], bits[32]), bits[1]) = invoke(tuple.42, to_apply=__for_1_func, id=43, pos=[(1,12,7)])
  tuple_index.44: (bits[64], bits[32]) = tuple_index(invoke.43, index=0, id=44, pos=[(1,12,7)])
  tuple_index.32: token = tuple_index(receive.31, index=0, id=32)
  tuple_index.45: bits[1] = tuple_index(invoke.43, index=1, id=45, pos=[(1,12,7)])
  tuple_index.47: bits[32] = tuple_index(tuple_index.44, index=1, id=47, pos=[(1,12,7)])
  tuple_index.48: bits[64] = tuple_index(tuple_index.44, index=0, id=48, pos=[(1,12,7)])
  tuple_index.34: bits[32] = tuple_index(tuple_index.33, index=1, id=34, pos=[(1,12,7)])
  tuple_index.35: bits[64] = tuple_index(tuple_index.33, index=0, id=35, pos=[(1,12,7)])
  send.46: token = send(tuple_index.32, tuple_index.44, predicate=tuple_index.45, channel_id=3, id=46, pos=[(1,12,7)])
  tuple.49: (bits[1], bits[32], bits[64]) = tuple(tuple_index.45, tuple_index.47, tuple_index.48, id=49)
  next (send.46, tuple.49)
}

proc foo_proc(tkn: token, st: (), init={()}) {
  literal.57: bits[32] = literal(value=0, id=57, pos=[(1,5,5)])
  literal.58: (bits[64], bits[32]) = literal(value=(0, 0), id=58, pos=[(1,5,5)])
  invoke.59: (bits[1], ((bits[64], bits[32]), bits[1]), bits[1], (bits[32], bits[1])) = invoke(literal.57, literal.58, to_apply=foo, id=59, pos=[(1,5,5)])
  in_pred: bits[1] = tuple_index(invoke.59, index=0, id=60, pos=[(1,7,15)])
  receive.61: (token, bits[32]) = receive(tkn, predicate=in_pred, channel_id=0, id=61, pos=[(1,7,15)])
  tuple_index.63: bits[32] = tuple_index(receive.61, index=1, id=63)
  invoke.64: (bits[1], ((bits[64], bits[32]), bits[1]), bits[1], (bits[32], bits[1])) = invoke(tuple_index.63, literal.58, to_apply=foo, id=64, pos=[(1,7,15)])
  __for_1_ctx_in_pred: bits[1] = tuple_index(invoke.64, index=2, id=69, pos=[(1,12,7)])
  receive.70: (token, (bits[64], bits[32])) = receive(tkn, predicate=__for_1_ctx_in_pred, channel_id=3, id=70, pos=[(1,12,7)])
  tuple_index.72: (bits[64], bits[32]) = tuple_index(receive.70, index=1, id=72)
  invoke.73: (bits[1], ((bits[64], bits[32]), bits[1]), bits[1], (bits[32], bits[1])) = invoke(tuple_index.63, tuple_index.72, to_apply=foo, id=73, pos=[(1,12,7)])
  tuple_index.65: ((bits[64], bits[32]), bits[1]) = tuple_index(invoke.64, index=1, id=65, pos=[(1,12,7)])
  tuple_index.74: (bits[32], bits[1]) = tuple_index(invoke.73, index=3, id=74, pos=[(1,20,7)])
  tuple_index.66: (bits[64], bits[32]) = tuple_index(tuple_index.65, index=0, id=66, pos=[(1,12,7)])
  __for_1_ctx_out_pred: bits[1] = tuple_index(tuple_index.65, index=1, id=67, pos=[(1,12,7)])
  tuple_index.75: bits[32] = tuple_index(tuple_index.74, index=0, id=75, pos=[(1,20,7)])
  out_pred: bits[1] = tuple_index(tuple_index.74, index=1, id=76, pos=[(1,20,7)])
  tuple_index.62: token = tuple_index(receive.61, index=0, id=62)
  send.68: token = send(tkn, tuple_index.66, predicate=__for_1_ctx_out_pred, channel_id=2, id=68, pos=[(1,12,7)])
  tuple_index.71: token = tuple_index(receive.70, index=0, id=71)
  send.77: token = send(tkn, tuple_index.75, predicate=out_pred, channel_id=1, id=77, pos=[(1,20,7)])
  after_all.78: token = after_all(tuple_index.62, send.68, tuple_index.71, send.77, id=78, pos=[(1,5,5)])
  tuple.79: () = tuple(id=79)
  next (after_all.78, tuple.79)
}
```

spurserh, 2022-09-22:
The IOs on the generated channels for the pipelined loop go through GenerateIOInvokes() like regular IO ops. This was changed to fan in/out, which then removed the token relationship between them. It's interesting that it worked at all up to now. I think xlscc needs a mechanism to specify explicit request/response patterns anyway. One of those constraints can then be specified for the pipelined loop request/response.

spurserh, 2022-09-26:
Is there an easy way to get a test to fail from this?

meheff, 2022-09-26:
Changing this line from `continue` to `break` should cause the test to fail:

https://github.com/google/xls/blob/5b88f30b812f64d2c67d5ec683c5eb322105fed0/xls/interpreter/proc_interpreter.cc#L192

With that change all tests pass except the xlscc translator test which fails due to deadlock.

spurserh, 2022-09-27:
That appears to be a pretty old version of proc_interpreter.cc . Is there an analogous change to make at HEAD?

meheff, 2022-09-29:
Sorry about that. I send you a suggested tweak, then submitted a change which completely changed the file. At head, if you change this line to include `Is<Send>` instead of `Is<Receive>` then the JIT will execute sends as late as possible which should trigger the deadlock:

https://github.com/google/xls/blob/e7745e03a35ef5f9a599b3eb26d03e871e7687ed/xls/interpreter/proc_interpreter.cc#L153

closed 2022-09-29 by spurserh