google/xls #727: DSLX: Instantiating a parametric type with a named conflict causes XlsTypeError

RobSpringer, 2022-09-29:
In one file, I have a parametric type,
```
pub struct SlidingWindow<NUM_ELEMENTS: u32> {
  data: Block[NUM_ELEMENTS],
  num_entries: u32,
}
```

in another file, I instantiate this type via
```
import sliding_window
type SlidingWindow = sliding_window::SlidingWindow<u32:4>;
```

This is fine.

But if I instead try
```
import sliding_window
const NUM_ENTRIES = u32:4;
type SlidingWindow = sliding_window::SlidingWindow<NUM_ENTRIES>;
```

I get the following error:
```
XlsTypeError: struct 'SlidingWindow' structure: SlidingWindow { data: uN[8][4][4][4], num_entries: uN[32] } vs struct 'SlidingWindow' structure: SlidingWindow { data: uN[8][4][4][NUM_ENTRIES], num_entries: uN[32] }: Return type of function body for 'init' did not match the annotated return type.
```
Looks like we're not resolving the symbolic constant to its final type at the point where we're doing this check.

rdob-ant, 2023-05-16:
I've ran into a similar issue today. My gut feeling is that it may have the same root cause as this one (constants sometimes not being resolved to their values when used as parameters of parametrics), so posting it here:
```
struct MyStruct<WIDTH: u32> {
    myfield: bits[WIDTH]
}

fn myfunc<FIELD_WIDTH: u32>(arg: MyStruct<FIELD_WIDTH>) -> u32 {
    (arg.myfield as u32)
}

const WIDTH_15 = u32: 15;

//
// These three are OK
//
fn myfunc_spec1(arg: MyStruct<u32: 15>) -> u32 {
    (myfunc<u32: 15>(arg))
}

fn myfunc_spec2(arg: MyStruct<u32: 15>) -> u32 {
    (myfunc<WIDTH_15>(arg))
}

fn myfunc_spec3(arg: MyStruct<u32: 15>) -> u32 {
    (myfunc(arg))
}

//
// Any of these three leads to an internal error in DSLX interpreter
//
fn myfunc_spec4(arg: MyStruct<WIDTH_15>) -> u32 {
    (myfunc<u32: 15>(arg))
}

fn myfunc_spec5(arg: MyStruct<WIDTH_15>) -> u32 {
    (myfunc<WIDTH_15>(arg))
}

fn myfunc_spec6(arg: MyStruct<WIDTH_15>) -> u32 {
    (myfunc(arg))
}
```

For above code, _interpreter_main_ produces following error:
```
I0516 15:17:00.812332       3 parametric_instantiator_internal.cc:115] Symbolically binding param 0 formal MyStruct { myfield: uN[FIELD_WIDTH] } against arg MyStruct { myfield: uN[WIDTH_15] }
E0516 15:17:00.837294       3 parametric_bind.cc:79] INTERNAL: XLS_RET_CHECK failure (xls/dslx/type_system/parametric_bind.cc:79) !arg_dim.IsParametric()
0x564c08dd9cc9: xabsl::StatusBuilder::CreateStatusAndConditionallyLog()
0x564c07203c33: xls::dslx::ParametricBindConcreteTypeDim()
0x564c07204b50: xls::dslx::ParametricBind()
0x564c07205120: xls::dslx::ParametricBind()
0x564c071fcca1: xls::dslx::internal::ParametricInstantiator::InstantiateOneArg()
0x564c07202481: xls::dslx::internal::FunctionInstantiator::Instantiate()
0x564c071fc80c: xls::dslx::InstantiateFunction()
0x564c071bdbb9: xls::dslx::CheckInvocation()
0x564c071afdf0: std::_Function_handler<>::_M_invoke()
0x564c071c34b5: xls::dslx::(anonymous namespace)::DeduceInstantiation()
0x564c071e2242: xls::dslx::(anonymous namespace)::DeduceInvocation()
0x564c071e36a6: xls::dslx::(anonymous namespace)::DeduceVisitor::HandleInvocation()
0x564c0730e9f2: xls::dslx::Invocation::Accept()
0x564c071cf813: xls::dslx::Deduce()
0x564c071afd8a: std::_Function_handler<>::_M_invoke()
0x564c07206d58: xls::dslx::DeduceCtx::Deduce()
0x564c071c15b7: xls::dslx::(anonymous namespace)::DeduceVisitor::HandleStatement()
0x564c0720d662: xls::dslx::Statement::Accept()
0x564c071cf813: xls::dslx::Deduce()
0x564c071afd8a: std::_Function_handler<>::_M_invoke()
0x564c07206d58: xls::dslx::DeduceCtx::Deduce()
0x564c071c18a1: xls::dslx::(anonymous namespace)::DeduceVisitor::HandleBlock()
0x564c0718bdfe: xls::dslx::Block::Accept()
0x564c071cf813: xls::dslx::Deduce()
0x564c071afd8a: std::_Function_handler<>::_M_invoke()
0x564c07206d58: xls::dslx::DeduceCtx::Deduce()
0x564c071c4d97: xls::dslx::DeduceAndResolve()
0x564c071b4e10: xls::dslx::CheckFunction()
0x564c071b7996: xls::dslx::CheckModule()
0x564c0714a628: xls::dslx::TypecheckModule()
0x564c0714c889: xls::dslx::ParseAndTypecheck()
0x564c07146b10: xls::dslx::ParseAndTest()
0x564c070ade34: main
```

CC @proppy

rdob-ant, 2023-05-22:
Hi @cdleary . This issue is not the same as #967 , but I think it overlaps a bit, as it's about propagation of `const` bindings to parametric expressions within struct type annotations, and correct evaluation of them. So perhaps you can also look into this in scope of #971 ?

And maybe you are already aware of some workarounds for this, except for avoiding use of `const`?

cdleary, 2023-05-22:
Thanks for the ping on this -- note that anything that says INTERNAL is a bug that should not be possible for a user to trigger. Taking a look...

cdleary, 2023-05-23:
In review, fix should land tomorrow. Thanks!

closed 2023-05-23 by copybara-service[bot]