google/xls #734: Failed test: //xls/fuzzer:run_fuzz_wide_codegen_test -- INVALID_ARGUMENT: Type annotation sN[108] is not a builtin bits type

tcal-x, 2022-10-04:
Following the install/build instructions on a Rodete system, at step `bazel test -c opt //...`, all tests passed except for:
```
//xls/fuzzer:run_fuzz_wide_codegen_test                   (48/50 cached) FAILED in 2 out of 50 in 16.0s
```
A failing log showed
```
exec ${PAGER:-/usr/bin/less} "$0" || exit 1
Executing tests from //xls/fuzzer:run_fuzz_wide_codegen_test
-----------------------------------------------------------------------------
Running tests under Python 3.10.7: /usr/bin/python3
[ RUN      ] RunFuzzTest.test_first_n_seeds_seed_3
[  FAILED  ] RunFuzzTest.test_first_n_seeds_seed_3
======================================================================
ERROR: test_first_n_seeds_seed_3 (__main__.RunFuzzTest)
RunFuzzTest.test_first_n_seeds_seed_3
test_first_n_seeds_seed_3(seed=3)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/usr/local/google/home/tcal/.cache/bazel/_bazel_tcal/67a734db6b602ae2774c3b311bac664d/sandbox/linux-sandbox/21776/execroot/com_google_xls/bazel-out/k8-opt/bin/xls/fuzzer/run_fuzz_wide_codegen_test.runfiles/com_google_absl_py/absl/testing/parameterized.py", line 282, in bound_param_test
    return test_method(self, **testcase_params)
  File "/usr/local/google/home/tcal/.cache/bazel/_bazel_tcal/67a734db6b602ae2774c3b311bac664d/sandbox/linux-sandbox/21776/execroot/com_google_xls/bazel-out/k8-opt/bin/xls/fuzzer/run_fuzz_wide_codegen_test.runfiles/com_google_xls/xls/fuzzer/run_fuzz_codegen_test.py", line 74, in test_first_n_seeds
    run_fuzz.generate_sample_and_run(
  File "/usr/local/google/home/tcal/.cache/bazel/_bazel_tcal/67a734db6b602ae2774c3b311bac664d/sandbox/linux-sandbox/21776/execroot/com_google_xls/bazel-out/k8-opt/bin/xls/fuzzer/run_fuzz_wide_codegen_test.runfiles/com_google_xls/xls/fuzzer/run_fuzz.py", line 280, in generate_sample_and_run
    smp = ast_generator.generate_sample(ast_generator_options, sample_options,
RuntimeError: INVALID_ARGUMENT: Type annotation sN[108] is not a builtin bits type

----------------------------------------------------------------------
Ran 1 test in 0.001s

FAILED (errors=1)
```

tcal-x, 2022-10-04:
This shows up in your nightly ubuntu tests: https://github.com/google/xls/actions/runs/3180113159/jobs/5183316790, so let me know if you want me to close this as redundant.

I'm at 6f977629dd2c843af74858479e18fcc198e35976.

meheff, 2022-10-04:
This should be fixed at https://github.com/google/xls/commit/3b2dd1125a512ea9b4d8f62e39d4155840254d99

tcal-x, 2022-10-04:
Confirmed after syncing and rerunning.

closed 2022-10-04 by tcal-x