google/xls #746: Support llvm runtime library in JIT interpreter to enable interpretation of operations like __udivti3 (or disable use of these operations).

hongted, 2022-10-19:
For a few  operations (ex. div by certain constants), LLVM JIT will instead rely on functions implemented within its runtime library (built-ins) like __udivti3.

This issue is to either 1) support the built-in operations; 2) force LLVM to not rely upon these operations; or 3) prevent a need for LLVM to use these operations by directly supporting code-gen/optimization of these operations down to XLS primitives.

hongted, 2022-10-19:
Example ir code:

```
package sample                                                                  
                                                                                
file_number 0 "fake_file.x"                                                     
                                                                                
top fn __sample__main(x0: bits[1], x1: bits[23], x2: bits[97], x3: bits[29]) -> bits[97] {
  literal.123: bits[1] = literal(value=0, id=123, pos=[(0,28,39)])              
  x5: bits[21] = literal(value=0, id=128, pos=[(0,9,20)])                       
  literal.13: bits[97] = literal(value=0xaaaa_aaaa_aaaa_aaaa_aaaa_aaaa, id=13, pos=[(0,11,34)])
  gate.50: bits[21] = gate(literal.123, x5, id=50, pos=[(0,28,22)])             
  ret x7: bits[97] = udiv(x2, literal.13, id=14, pos=[(0,11,24)])               
}             
```

hongted, 2022-10-19:
Errors out with `JIT session error: Symbols not found: [ __udivti3 ]`

grebe, 2024-03-26:
This was mentioned as an issue today