google/xls #750: Improve measurement of evaluation time in benchmark_main

meheff, 2022-10-21:
benchmark_main (https://github.com/google/xls/blob/main/xls/tools/benchmark_main.cc) measures the evaluation time of a function/proc using the interpreter and jit. Right now it only works for functions (#749) and just runs a fixed number of iterations (fixed for all benchmarks). That number is often too small and results in run times of only a few ms which is very noisy. It'd be better to run the function/proc for some amount of time and report the rate (e.g, iterations per second).

vincent-mirian-google, 2022-10-29:
Does https://github.com/google/xls/commit/ab5c39030cf5802d58dfb10c619af81cb9be2d7d tackle this?

meheff, 2022-10-29:
yeah, it's fixed now.

closed 2022-10-29 by meheff