google/xls #762: Remove constant literals in boolean expression in generated Verilog (codegen)

vincent-mirian-google, 2022-10-31:
The following IR:
```
package sample

file_number 0 "fake_file.x"

chan sample__operand_0(bits[32], id=0, kind=streaming, ops=receive_only, flow_control=ready_valid, metadata="""""")
chan sample__operand_1(bits[32], id=1, kind=streaming, ops=receive_only, flow_control=ready_valid, metadata="""""")
chan sample__result(bits[32], id=2, kind=streaming, ops=send_only, flow_control=ready_valid, metadata="""""")

top proc __sample__main_0_next(__token: token, init={}) {
  receive.4: (token, bits[32]) = receive(__token, channel_id=0, id=4)
  receive.7: (token, bits[32]) = receive(__token, channel_id=1, id=7)
  tok_operand_0_val: token = tuple_index(receive.4, index=0, id=5, pos=[(0,14,9)])
  tok_operand_1_val: token = tuple_index(receive.7, index=0, id=8, pos=[(0,15,9)])
  operand_0_val: bits[32] = tuple_index(receive.4, index=1, id=6, pos=[(0,14,28)])
  operand_1_val: bits[32] = tuple_index(receive.7, index=1, id=9, pos=[(0,15,28)])
  tok_recv: token = after_all(tok_operand_0_val, tok_operand_1_val, id=10)
  result_val: bits[32] = add(operand_0_val, operand_1_val, id=11, pos=[(0,18,35)])
  tok_send: token = send(tok_recv, result_val, channel_id=2, id=12)
  after_all.14: token = after_all(__token, tok_operand_0_val, tok_operand_1_val, tok_recv, tok_send, id=14)
  next (after_all.14)
}
```

using the following command:
```
xls/tools/codegen_main --output_signature_path=module_sig.textproto --delay_model=unit --generator=combinational sample.opt.ir --logtostderr
```

produces the following Verilog:
```
module __sample__main_0_next(
  input wire [31:0] sample__operand_0,
  input wire sample__operand_0_vld,
  input wire [31:0] sample__operand_1,
  input wire sample__operand_1_vld,
  input wire sample__result_rdy,
  output wire [31:0] sample__result,
  output wire sample__result_vld,
  output wire sample__operand_0_rdy,
  output wire sample__operand_1_rdy
);
  wire [31:0] result_val;
  wire and_35;
  assign result_val = sample__operand_0 + sample__operand_1;
  assign and_35 = sample__result_rdy;
  assign sample__result = result_val;
  assign sample__result_vld = sample__operand_0_vld & sample__operand_1_vld & 1'h1 & 1'h1;
  assign sample__operand_0_rdy = and_35;
  assign sample__operand_1_rdy = and_35;
endmodule
```

For
```
  assign sample__result_vld = sample__operand_0_vld & sample__operand_1_vld & 1'h1 & 1'h1;
```
is equivalent to 
```
  assign sample__result_vld = sample__operand_0_vld & sample__operand_1_vld;
```

Remove the unnecessary constant literals.

cdleary, 2022-10-31:
I guess to do this means we'd need opportunistic constant folding in the codegen pass pipeline since I think this gets introduced when we're producing the I/O signals. Maybe @meheff has general thoughts on long term for potential optimizations in codegen transforms though? To do it the local rewrite way seems bounded complexity/difficulty though.

meheff, 2022-11-01:
I think having a few simplification/cleanpu passes in codegen makes sense. Could be constant prop and some logical optimizations. I guess in theory we could run any optimization pass, but we'd want to curate them to make sure they don't increase critical path. This might require carving out some of the optimizations in arith_simplification and others into separate passes have that property (doesn't increate critical path).

commit d3d2e8cc (2024-03-18): Split arithmetic simplification pass into two separate passes.
Arithmetic simplification is a grab bag of pattern matching optimizations. Split the pass into two parts: one for trivial, no-brainer pattern optimizations call "basic_simplification_pass", and everything else (more complicated arithmetic transformations). I imagine more optimizations could be moved in to the basic simplification pass and this pass could be run more frequently.

Also, add the "basic" simplification pass to the codegen pipeline to clean up the generated flow control logic. Example simplification:

  wire p0_all_acti [... cut]
closed 2024-03-18 by copybara-service[bot]