google/xls #763: Add waveform traces output to testbench to simulator

vincent-mirian-google, 2022-11-01:
Currently, XLS supports iverilog for simulating verilog module with a testbench. When simulating iverilog, the waveform of the testbench can be created for viewing by other open source tools (e.g. [gtkwave](https://gtkwave.sourceforge.net/)).

To produce the waveform traces, the following can be added to the initial block of the testbench.
```
  $dumpfile("some_file_name.vcd");
  $dumpvars(0, test_bench_module_name);
```

Then, viewed by gtkwave:
```
gtkwave some_file_name.vcd
```

Forchapeatl, 2022-11-02:
@vincent-mirian-google please may I take this issue ?

vincent-mirian-google, 2022-11-02:
Hi @Forchapeatl You are welcome to tackle this issue. I assigned it to you. Thank you.

Forchapeatl, 2022-11-03:
> ```
>   $dumpfile("some_file_name.vcd");
>   $dumpvars(0, test_bench_module_name);
> ```
> 
> Then, viewed by gtkwave:
> 
> ```
> gtkwave some_file_name.vcd
> ```

Please are we to include it in these two files

-  [xls/uncore_rtl/ice40/wrap_io_ice40_repeat_byte_4_testbench.v](https://github.com/google/xls/blob/main/xls/uncore_rtl/ice40/wrap_io_ice40_repeat_byte_4_testbench.v)
- [xls/uncore_rtl/ice40/wrap_io_ice40_testbench.v](https://github.com/google/xls/blob/main/xls/uncore_rtl/ice40/wrap_io_ice40_testbench.v)

vincent-mirian-google, 2022-11-03:
These files do not need to change. The GH issue is to programatically update the generated testbench (found in https://github.com/google/xls/blob/main/xls/simulation/module_testbench.h#L253) to output the waveform trace file.