google/xls #766: How to handling input arrays (external memories)

xls-interested, 2022-11-03:
void test_input_array(int a[4], int n) {
  for (int i = 0; i < 4; i++) {
    a[i] = a[i] + 5;
  }
}

gives

Generating IR...
UNIMPLEMENTED: For loop missing #pragma
 --> ./test_input_array.cpp:2:3
  |
2 |   for (int i = 0; i < 4; i++) {
  |   ^

meheff, 2022-11-07:
We are currently working on improving our external memory support. I'm not too sure how these would map up to the C++ level in xlscc. @spurserh might know more.

grebe, 2022-11-29:
4f0aa2821bf8241e780dc27157c22723a5f72892 added a codegen option to support external memories, see [delay.x](https://github.com/google/xls/blob/main/xls/examples/delay.x) (and its [BUILD](https://github.com/google/xls/blob/9945042db2acc055c8f5a554d9a4a021fb55b17a/xls/examples/BUILD#L480)) for an example usage in DSLX. Currently, I think it will only work from C++ by using `ac_channel`, but we'd like array accesses like what you wrote to be ultimately lowered to similar IR that can be codegen'd for external memories. @spurserh can comment more on what that would look like.