google/xls #770: DSLX code passes parse and typechcking fails in ir conversion/interpretation

vincent-mirian-google, 2022-11-03:
The following code snippet parsing and type checks.
```
proc main {
  x12: chan<u8> in;
  config(x27: chan<u8> in) {
    (x12,)
  }
  next(x0: token) {
    ()
  }
}
```
However, it returns an error when running it through the DSLX interpreter:
```
I1103 07:49:25.834760 2806348 proc_config_ir_converter.cc:159] ProcConfigIrConverter::HandleNameRef : x12
F1103 07:49:25.834769 2806348 throw_delegate.cc:121] RAW: absl::container_internal::raw_hash_map<>::at
```
When it is changed to the following, the error is no longer present:
```
proc main {
  x12: chan<u8> in;
  config(x12: chan<u8> in) {
    (x12,)
  }
}
```

All the examples that I have seen have the config names in the param list match the proc param list. Is this a requirement? If so, maybe we can have a better error msg?

vincent-mirian-google, 2022-11-03:
Spoke to @RobSpringer . He is looking into it. Thanks 👍

RobSpringer, 2022-11-03:
Fixed by above commit.

closed 2022-11-03 by RobSpringer