google/xls #773: how to generate compile_command.json for this project

KuiWei004, 2022-11-08:
I want to use VSCode with clangd, but there is not compile_command.json generated after building.
I tried to use _Hedron's Compile Commands Extractor for Bazel_, but it didn't work. Maybe I take the wrong way to use this tool.

meheff, 2022-11-08:
I don't know of anyone who has done this before. After Googling. It looks like there are at least a few different approaches:

* https://github.com/grailbio/bazel-compilation-database
* https://github.com/hedronvision/bazel-compile-commands-extractor
* https://github.com/stackb/bazel-stack-vscode-cc

I haven't tried any, but if you (or anyone else) has success please post back.

cdleary, 2023-08-21:
@hzeller added a script that does this in https://github.com/google/xls/commit/da47c25e7e65c64e87d1f202210dba96c10c3b6d -- the path is now https://github.com/google/xls/blob/main/dev_utils/make-compilation-db.sh

closed 2023-08-21 by cdleary