google/xls #774: IR Conversion failure

vincent-mirian-google, 2022-11-08:
Using the cmd:
```
bazel run -c opt //xls/dslx:ir_converter_main -- --top=main --logtostderr sample.x
```

with the following DSLX code:
```
type x1 = token;
proc main {
  config() {
    ()
  }
  next(x0: token) {
    let x2: x1[1] = [x0];
    let x3: u13 = u13:0xaaa;
    let x4: u13 = (x3) / (u13:0x1fff);
    let x5: u13 = (x3)[x3+:u13];
    let x6: u13 = bit_slice_update(x3, x4, x5);
    let x7: u13 = -(x5);
    let x8: u13 = one_hot_sel(u3:0x0, [x3, x7, x7]);
    let x9: token = join(x0);
    let x10: u13 = !(x8);
    let x11: u13 = gate!((x8) >= (x8), x3);
    let x12: u13 = (x3) - (x6);
    let x13: u13 = x10;
    let x14: u3 = (x12)[1:4];
    let x15: u13 = (x8) + (x11);
    let x16: u13 = !(x12);
    let x17: token = join(x9, x0);
    let x18: x1 = (x2)[if (x16) >= (u13:0) { u13:0 } else { x16 }];
    let x19: u39 = ((x7) ++ (x10)) ++ (x15);
    let x20: u13 = (x15) ^ (x6);
    let x21: u3 = for (i, x): (u4, u3) in u4:0..u4:1 {
      x
    }(x14);
    let x22: u13 = (x20) >> (if (x15) >= (u13:12) { u13:12 } else { x15 });
    let x23: u13 = (x6) | (((x5) as u13));
    let x24: x1[1] = update(x2, if (x21) >= (u3:0) { u3:0 } else { x21 }, x18);
    let x25: bool = and_reduce(x21);
    ()
  }
}
```

received:
```
*** SIGSEGV (@0x20) ... ***
PC: @     0x55682ed3ae83  (unknown)  xls::BValue::GetType()
    @     0x55682efd917a       1152  FailureSignalHandler()
    @     0x7fec8e18d1c0  (unknown)  (unknown)
    @     0x55682ed3ae83          8  xls::BValue::GetType()
    @     0x55682ea7720e        336  xls::dslx::(anonymous namespace)::FunctionConverterVisitor::HandleLet()
    @     0x55682ec0201b         32  xls::dslx::Let::Accept()
    @     0x55682ea701a4        464  xls::dslx::(anonymous namespace)::FunctionConverterVisitor::Visit()
    @     0x55682ea772d7        336  xls::dslx::(anonymous namespace)::FunctionConverterVisitor::HandleLet()
    @     0x55682ec0201b         32  xls::dslx::Let::Accept()
    @     0x55682ea701a4        464  xls::dslx::(anonymous namespace)::FunctionConverterVisitor::Visit()
    @     0x55682ea772d7        336  xls::dslx::(anonymous namespace)::FunctionConverterVisitor::HandleLet()
    @     0x55682ec0201b         32  xls::dslx::Let::Accept()
    @     0x55682ea701a4        464  xls::dslx::(anonymous namespace)::FunctionConverterVisitor::Visit()
    @     0x55682ea772d7        336  xls::dslx::(anonymous namespace)::FunctionConverterVisitor::HandleLet()
    @     0x55682ec0201b         32  xls::dslx::Let::Accept()
    @     0x55682ea701a4        464  xls::dslx::(anonymous namespace)::FunctionConverterVisitor::Visit()
    @     0x55682ea772d7        336  xls::dslx::(anonymous namespace)::FunctionConverterVisitor::HandleLet()
    @     0x55682ec0201b         32  xls::dslx::Let::Accept()
    @     0x55682ea701a4        464  xls::dslx::(anonymous namespace)::FunctionConverterVisitor::Visit()
    @     0x55682ea772d7        336  xls::dslx::(anonymous namespace)::FunctionConverterVisitor::HandleLet()
    @     0x55682ec0201b         32  xls::dslx::Let::Accept()
    @     0x55682ea701a4        464  xls::dslx::(anonymous namespace)::FunctionConverterVisitor::Visit()
    @     0x55682ea772d7        336  xls::dslx::(anonymous namespace)::FunctionConverterVisitor::HandleLet()
    @     0x55682ec0201b         32  xls::dslx::Let::Accept()
    @     0x55682ea701a4        464  xls::dslx::(anonymous namespace)::FunctionConverterVisitor::Visit()
    @     0x55682ea772d7        336  xls::dslx::(anonymous namespace)::FunctionConverterVisitor::HandleLet()
    @     0x55682ec0201b         32  xls::dslx::Let::Accept()
    @     0x55682ea701a4        464  xls::dslx::(anonymous namespace)::FunctionConverterVisitor::Visit()
    @     0x55682ea772d7        336  xls::dslx::(anonymous namespace)::FunctionConverterVisitor::HandleLet()
    @     0x55682ec0201b         32  xls::dslx::Let::Accept()
    @     0x55682ea701a4        464  xls::dslx::(anonymous namespace)::FunctionConverterVisitor::Visit()
    @     0x55682ea772d7        336  xls::dslx::(anonymous namespace)::FunctionConverterVisitor::HandleLet()
    @     0x55682ec0201b         32  xls::dslx::Let::Accept()
    @     0x55682ea701a4        464  xls::dslx::(anonymous namespace)::FunctionConverterVisitor::Visit()
    @     0x55682ea772d7        336  xls::dslx::(anonymous namespace)::FunctionConverterVisitor::HandleLet()
    @     0x55682ec0201b         32  xls::dslx::Let::Accept()
    @     0x55682ea701a4        464  xls::dslx::(anonymous namespace)::FunctionConverterVisitor::Visit()
    @     0x55682ea772d7        336  xls::dslx::(anonymous namespace)::FunctionConverterVisitor::HandleLet()
    @     0x55682ec0201b         32  xls::dslx::Let::Accept()
    @     0x55682ea701a4        464  xls::dslx::(anonymous namespace)::FunctionConverterVisitor::Visit()
    @     0x55682ea772d7        336  xls::dslx::(anonymous namespace)::FunctionConverterVisitor::HandleLet()
    @     0x55682ec0201b         32  xls::dslx::Let::Accept()
    @     0x55682ea701a4        464  xls::dslx::(anonymous namespace)::FunctionConverterVisitor::Visit()
    @     0x55682ea772d7        336  xls::dslx::(anonymous namespace)::FunctionConverterVisitor::HandleLet()
    @     0x55682ec0201b         32  xls::dslx::Let::Accept()
    @     0x55682ea701a4        464  xls::dslx::(anonymous namespace)::FunctionConverterVisitor::Visit()
    @     0x55682ea772d7        336  xls::dslx::(anonymous namespace)::FunctionConverterVisitor::HandleLet()
    @     0x55682ec0201b         32  xls::dslx::Let::Accept()
    @     0x55682ea701a4        464  xls::dslx::(anonymous namespace)::FunctionConverterVisitor::Visit()
    @     0x55682ea772d7        336  xls::dslx::(anonymous namespace)::FunctionConverterVisitor::HandleLet()
    @     0x55682ec0201b         32  xls::dslx::Let::Accept()
    @     0x55682ea701a4        464  xls::dslx::(anonymous namespace)::FunctionConverterVisitor::Visit()
    @     0x55682ea772d7        336  xls::dslx::(anonymous namespace)::FunctionConverterVisitor::HandleLet()
    @     0x55682ec0201b         32  xls::dslx::Let::Accept()
    @     0x55682ea701a4        464  xls::dslx::(anonymous namespace)::FunctionConverterVisitor::Visit()
    @     0x55682ea71cb7        144  xls::dslx::(anonymous namespace)::FunctionConverterVisitor::HandleBlock()
    @     0x55682ebfc15b         32  xls::dslx::Block::Accept()
    @     0x55682ea701a4        464  xls::dslx::(anonymous namespace)::FunctionConverterVisitor::Visit()
    @     0x55682ea66166       1664  xls::dslx::(anonymous namespace)::ConvertOneFunctionInternal()
    @     0x55682ea61783        848  xls::dslx::ConvertCallGraph()
    @     0x55682ea62a02        320  xls::dslx::ConvertOneFunctionIntoPackageInternal<>()
    @     0x55682ea620fe        256  xls::dslx::ConvertOneFunctionIntoPackage()
    @     0x55682ea57745       1968  main
    @     0x7fec8e020633        192  __libc_start_main
    @ ... and at least 1 more frames
```

vincent-mirian-google, 2022-11-08:
This may be related to https://github.com/google/xls/issues/772.

juttadegener, 2023-06-26:
A smaller recent version of this:
```
proc main {
  init {}
  config() {}
  next(t: token, st:()) {
    let x: token = [t][u1:0];
    ()
  }
}
```
The key trick is to somehow smuggle a token into an array.
You can't write it directly, but some indirect ways work.

closed 2024-04-19 by copybara-service[bot]