google/xls #783: opt_main segfaults in z3 on proc input

meheff, 2022-11-14:
Looks like it has something to do with z3 during the mutual exclusion pass. This call to Z3 is segfaulting:

https://github.com/google/xls/blob/288bc7378c8731cf08f7a1facba96101d6df0590/xls/passes/mutual_exclusion_pass.cc#L246

This does not repro with the google-internal version of z3. Looks like we use v4.8.8 internally, and v4.8.7 in github. Updating to 4.8.8 might be a reasonable first stab at fixing this.

Repro:

```
bazel-bin/xls/opt_main bad.ir
```

bad.ir:

```
package sample

file_number 0 "fake_file.x"

chan sample__x22(bits[45], id=0, kind=streaming, ops=receive_only, flow_control=ready_valid, metadata="""""")

top proc __sample__main_0_next(__token: token, __state: bits[63], init={4611686018427387903}) {
  x5: bits[64] = one_hot(__state, lsb_prio=false, id=8, pos=[(0,14,25)])
  x6: bits[64] = neg(x5, id=9, pos=[(0,15,18)])
  x10: bits[1] = xor_reduce(x6, id=14)
  x23: (token, bits[45]) = receive(__token, predicate=x10, channel_id=0, id=34)
  tuple_index.35: token = tuple_index(x23, index=0, id=35)
  after_all.42: token = after_all(__token, tuple_index.35, id=42)
  next (after_all.42, __state)
}
```

meheff, 2022-11-14:
Maybe relevant final log message before blowing up:

```
I1114 13:51:36.838349  807540 abstract_node_evaluator.h:39] Handling x10: bits[1] = xor_reduce(x6, id=14)
I1114 13:51:36.838400  807540 z3_ir_translator.cc:1284] Unhandled node for conversion from XLS IR to Z3, defaulting to variable: %sx23: (token, bits[45]) = receive(__token, predicate=x10, channel_id=0, id=34)
Segmentation fault
```

meheff, 2022-11-14:
Logs above were for opt mode. Might be worthwhile building and running in non-opt as it could enable assertions which may be enlightening.

taktoa, 2022-11-14:
https://github.com/Z3Prover/z3/releases/tag/z3-4.8.8 there are a couple assertion violation bugs fixed in that release, but they both have to do with maxsat, which we're not using (except in the Z3 graph coloring implementation, which we don't use other than in the graph coloring tests). So I'm fairly skeptical that updating to 4.8.8 is going to change much.

vincent-mirian-google, 2022-11-14:
Another one that has the same issue.
```
package sample

file_number 0 "fake_file.x"

chan sample__x12(bits[34], id=0, kind=streaming, ops=receive_only, flow_control=ready_valid, metadata="""""")

top proc __sample__main_0_next(__token: token, __state: bits[24], init={11184810}) {
  literal.5: bits[24] = literal(value=18, id=5, pos=[(0,10,43)])
  uge.6: bits[1] = uge(__state, literal.5, id=6, pos=[(0,10,35)])
  literal.7: bits[24] = literal(value=18, id=7, pos=[(0,10,53)])
  sel.8: bits[24] = sel(uge.6, cases=[__state, literal.7], id=8, pos=[(0,10,27)])
  x3: bits[24] = shrl(__state, sel.8, id=9, pos=[(0,10,23)])
  x5: bits[3] = literal(value=2, id=11, pos=[(0,12,20)])
  x4: bits[24] = not(x3, id=10, pos=[(0,11,18)])
  x22: bits[3] = dynamic_bit_slice(x5, x4, width=3, id=33, pos=[(0,28,22)])
  literal.39: bits[3] = literal(value=0, id=39, pos=[(0,32,44)])
  x7: bits[1] = and_reduce(x3, id=13)
  uge.40: bits[1] = uge(x22, literal.39, id=40, pos=[(0,32,37)])
  literal.41: bits[3] = literal(value=0, id=41, pos=[(0,32,52)])
  x9: bits[1] = or(x7, x7, id=15, pos=[(0,16,24)])
  x2: bits[24] = not(__state, id=4, pos=[(0,9,18)])
  x13: (token, bits[34]) = receive(__token, predicate=x7, channel_id=0, id=18)
  sel.42: bits[3] = sel(uge.40, cases=[x22, literal.41], id=42, pos=[(0,32,28)])
  bit_slice.31: bits[1] = bit_slice(x5, start=0, width=1, id=31)
  x17: bits[1] = dynamic_bit_slice(x9, __state, width=1, id=26, pos=[(0,23,24)])
  x6: bits[24] = bit_slice_update(__state, x2, x2, id=12, pos=[(0,13,34)])
  x18: bits[27] = literal(value=256, id=27, pos=[(0,24,23)])
  x15: bits[34] = tuple_index(x13, index=1, id=23, pos=[(0,21,22)])
  x26: bits[24] = shrl(__state, sel.42, id=43, pos=[(0,32,24)])
  x21: bits[1] = and(x9, bit_slice.31, id=32, pos=[(0,27,25)])
  zero_ext.35: bits[1] = zero_ext(x17, new_bit_count=1, id=35)
  literal.24: bits[24] = literal(value=15, id=24, pos=[(0,22,23)])
  x8: bits[24] = not(x6, id=14, pos=[(0,15,18)])
  bit_slice.29: bits[24] = bit_slice(x18, start=0, width=24, id=29)
  literal.37: bits[24] = literal(value=0, id=37, pos=[(0,31,24)])
  x23: bits[1] = or_reduce(x15, id=34)
  bit_slice.46: bits[1] = bit_slice(x26, start=0, width=1, id=46)
  x24: bits[1] = sub(x21, zero_ext.35, id=36, pos=[(0,30,26)])
  x30: bits[49] = literal(value=0, id=48, pos=[(0,36,23)])
  tuple_index.19: token = tuple_index(x13, index=0, id=19)
  x19: token = after_all(__token, id=28)
  literal.3: bits[1] = literal(value=1, id=3)
  x10: bits[24] = add(x2, __state, id=16, pos=[(0,17,24)])
  x11: bits[3] = not(x5, id=17, pos=[(0,18,18)])
  literal.20: bits[32] = literal(value=0, id=20, pos=[(0,20,25)])
  x14: token = tuple_index(x13, index=0, id=21, pos=[(0,20,24)])
  literal.22: bits[32] = literal(value=1, id=22, pos=[(0,21,23)])
  x16: bits[9] = dynamic_bit_slice(x2, literal.24, width=9, id=25, pos=[(0,22,22)])
  x20: bits[24] = or(x8, bit_slice.29, id=30, pos=[(0,26,24)])
  x25: bits[19] = dynamic_bit_slice(__state, literal.37, width=19, id=38, pos=[(0,31,23)])
  x27: bits[24] = sub(x6, x8, id=44, pos=[(0,33,24)])
  x28: bits[24] = neg(x2, id=45, pos=[(0,34,19)])
  x29: bits[1] = ult(x23, bit_slice.46, id=47, pos=[(0,35,26)])
  x31: bits[1] = neg(x24, id=49, pos=[(0,37,20)])
  x32: bits[1] = and_reduce(x30, id=50)
  after_all.51: token = after_all(__token, tuple_index.19, x19, id=51)
  next (after_all.51, __state)
}
```

vincent-mirian-google, 2022-11-14:
Another sample:
```
package sample

file_number 0 "fake_file.x"

chan sample__x23(bits[42], id=0, kind=streaming, ops=receive_only, flow_control=ready_valid, metadata="""""")

fn ____sample__main_0_next_counted_for_0_body(i: bits[4], __token_wrapped: (token, bits[1], bits[61])) -> (token, bits[1], bits[61]) {
  literal.33: bits[4] = literal(value=0, id=33)
  tuple_index.36: token = tuple_index(__token_wrapped, index=0, id=36)
  tuple_index.37: bits[1] = tuple_index(__token_wrapped, index=1, id=37)
  tuple_index.38: bits[61] = tuple_index(__token_wrapped, index=2, id=38)
  add.34: bits[4] = add(i, literal.33, id=34)
  ret tuple.39: (token, bits[1], bits[61]) = tuple(tuple_index.36, tuple_index.37, tuple_index.38, id=39)
}

top proc __sample__main_0_next(__token: token, __state: bits[61], init={870151226081154345}) {
  reverse.4: bits[61] = reverse(__state, id=4, pos=[(0,11,21)])
  one_hot.5: bits[62] = one_hot(reverse.4, lsb_prio=true, id=5, pos=[(0,11,21)])
  encode.6: bits[6] = encode(one_hot.5, id=6, pos=[(0,11,21)])
  x2: bits[61] = zero_ext(encode.6, new_bit_count=61, id=7, pos=[(0,11,21)])
  literal.11: bits[61] = literal(value=11, id=11, pos=[(0,14,43)])
  uge.12: bits[1] = uge(x2, literal.11, id=12, pos=[(0,14,35)])
  literal.13: bits[61] = literal(value=11, id=13, pos=[(0,14,53)])
  sel.14: bits[61] = sel(uge.12, cases=[x2, literal.13], id=14, pos=[(0,14,27)])
  x5: bits[61] = shrl(__state, sel.14, id=15, pos=[(0,14,23)])
  x4: bits[1] = xor_reduce(x2, id=10)
  literal.3: bits[1] = literal(value=1, id=3)
  x7: bits[61] = neg(x5, id=19, pos=[(0,16,18)])
  bit_slice.44: bits[1] = bit_slice(x4, start=0, width=1, id=44)
  tuple.40: (token, bits[1], bits[61]) = tuple(__token, literal.3, x7, id=40)
  zero_ext.16: bits[61] = zero_ext(__state, new_bit_count=61, id=16)
  x21: bits[1][1] = array(bit_slice.44, id=45, pos=[(0,31,24)])
  counted_for.41: (token, bits[1], bits[61]) = counted_for(tuple.40, trip_count=2, stride=1, body=____sample__main_0_next_counted_for_0_body
, id=41)
  ule.17: bits[1] = ule(x2, zero_ext.16, id=17, pos=[(0,15,29)])
  x28: bits[1] = ne(x21, x21, id=60, pos=[(0,37,26)])
  x19: bits[61] = tuple_index(counted_for.41, index=2, id=43)
  x6: bits[61] = gate(ule.17, x5, id=18, pos=[(0,15,23)])
  zero_ext.63: bits[61] = zero_ext(x28, new_bit_count=61, id=63)
  reverse.46: bits[61] = reverse(x19, id=46, pos=[(0,32,22)])
  reverse.56: bits[61] = reverse(x19, id=56, pos=[(0,36,22)])
  x15: bits[1] = and_reduce(x6, id=27)
  zero_ext.62: bits[61] = zero_ext(x19, new_bit_count=61, id=62)
  zero_ext.64: bits[61] = zero_ext(zero_ext.63, new_bit_count=61, id=64)
  x16: bits[1] = and_reduce(x5, id=28)
  one_hot.47: bits[62] = one_hot(reverse.46, lsb_prio=true, id=47, pos=[(0,32,22)])
  one_hot.57: bits[62] = one_hot(reverse.56, lsb_prio=true, id=57, pos=[(0,36,22)])
  zero_ext.29: bits[61] = zero_ext(x15, new_bit_count=61, id=29)
  x31: (bits[61], bits[61]) = smulp(zero_ext.62, zero_ext.64, id=65)
  x24: (token, bits[42]) = receive(__token, predicate=x16, channel_id=0, id=50)
  literal.8: bits[61] = literal(value=37, id=8, pos=[(0,12,23)])
  x12: bits[61] = neg(x6, id=24, pos=[(0,21,19)])
  encode.48: bits[6] = encode(one_hot.47, id=48, pos=[(0,32,22)])
  encode.58: bits[6] = encode(one_hot.57, id=58, pos=[(0,36,22)])
  x17: bits[1] = ugt(zero_ext.29, x5, id=30, pos=[(0,26,37)])
  tuple_index.67: bits[61] = tuple_index(x31, index=0, id=67, pos=[(0,41,10)])
  tuple_index.69: bits[61] = tuple_index(x31, index=1, id=69, pos=[(0,41,20)])
  x9: () = tuple(id=21, pos=[(0,18,17)])
  tuple_index.51: token = tuple_index(x24, index=0, id=51)
  x3: bits[24] = dynamic_bit_slice(__state, literal.8, width=24, id=9, pos=[(0,12,22)])
  x8: bits[1] = ult(x5, x7, id=20, pos=[(0,17,24)])
  x10: bits[61] = or(__state, x7, id=22, pos=[(0,19,24)])
  x11: bits[61] = neg(__state, id=23, pos=[(0,20,19)])
  x13: bits[1] = xor_reduce(x2, id=25)
  x14: bits[42] = literal(value=2199023255551, id=26, pos=[(0,23,23)])
  x18: bits[61] = dynamic_bit_slice(__state, x12, width=61, id=31, pos=[(0,27,23)])
  tuple_index.42: token = tuple_index(counted_for.41, index=0, id=42)
  x22: bits[61] = zero_ext(encode.48, new_bit_count=61, id=49, pos=[(0,32,22)])
  literal.52: bits[32] = literal(value=0, id=52, pos=[(0,34,25)])
  x25: token = tuple_index(x24, index=0, id=53, pos=[(0,34,24)])
  literal.54: bits[32] = literal(value=1, id=54, pos=[(0,35,23)])
  x26: bits[42] = tuple_index(x24, index=1, id=55, pos=[(0,35,22)])
  x27: bits[61] = zero_ext(encode.58, new_bit_count=61, id=59, pos=[(0,36,22)])
  x30: bits[1][1] = array(x17, id=61, pos=[(0,38,22)])
  literal.66: bits[32] = literal(value=0, id=66, pos=[(0,41,11)])
  literal.68: bits[32] = literal(value=1, id=68, pos=[(0,41,21)])
  x32: bits[61] = add(tuple_index.67, tuple_index.69, id=70, pos=[(0,41,14)])
  x33: bits[1] = ne(x9, x9, id=71, pos=[(0,43,25)])
  after_all.72: token = after_all(__token, tuple_index.51, id=72)
  next (after_all.72, x6)
}

```

vincent-mirian-google, 2022-11-14:
Another sample:
```
```

vincent-mirian-google, 2022-11-14:
Another sample:
```
package sample

file_number 0 "fake_file.x"

chan sample__x19(bits[47], id=0, kind=streaming, ops=receive_only, flow_control=ready_valid, metadata="""""")

top proc __sample__main_0_next(__token: token, __state: bits[33], init={0}) {
  x2: bits[34] = one_hot(__state, lsb_prio=true, id=5, pos=[(0,10,25)])
  x5: bits[6] = bit_slice(x2, start=28, width=6, id=8, pos=[(0,13,21)])
  x6: bits[1] = or_reduce(x5, id=9)
  x8: bits[1] = bit_slice(x6, start=0, width=1, id=12, pos=[(0,16,23)])
  x3: bits[34] = neg(x2, id=6, pos=[(0,11,18)])
  zero_ext.13: bits[6] = zero_ext(x8, new_bit_count=6, id=13)
  x4: bits[6] = dynamic_bit_slice(x3, x3, width=6, id=7, pos=[(0,12,21)])
  zero_ext.14: bits[6] = zero_ext(zero_ext.13, new_bit_count=6, id=14)
  zero_ext.15: bits[6] = zero_ext(x4, new_bit_count=6, id=15)
  x9: (bits[6], bits[6]) = smulp(zero_ext.14, zero_ext.15, id=16)
  tuple_index.18: bits[6] = tuple_index(x9, index=0, id=18, pos=[(0,19,9)])
  tuple_index.20: bits[6] = tuple_index(x9, index=1, id=20, pos=[(0,19,18)])
  literal.10: bits[6] = literal(value=42, id=10, pos=[(0,15,28)])
  x10: bits[6] = add(tuple_index.18, tuple_index.20, id=21, pos=[(0,19,13)])
  x7: bits[6] = udiv(x5, literal.10, id=11, pos=[(0,15,22)])
  literal.45: bits[34] = literal(value=0, id=45, pos=[(0,35,41)])
  sign_ext.25: bits[6] = sign_ext(x10, new_bit_count=6, id=25)
  literal.26: bits[6] = literal(value=0, id=26, pos=[(0,24,34)])
  x13: (bits[33], bits[6], bits[6]) = tuple(__state, x7, x4, id=24, pos=[(0,23,29)])
  uge.46: bits[1] = uge(x2, literal.45, id=46, pos=[(0,35,33)])
  literal.47: bits[34] = literal(value=0, id=47, pos=[(0,35,50)])
  literal.50: bits[6] = literal(value=0, id=50, pos=[(0,36,49)])
  literal.55: bits[6] = literal(value=0, id=55, pos=[(0,37,23)])
  x17: bits[1] = neg(x6, id=30, pos=[(0,27,20)])
  x14: bits[6] = dynamic_bit_slice(sign_ext.25, literal.26, width=6, id=27, pos=[(0,24,33)])
  x26: (bits[33], bits[6], bits[6])[1] = array(x13, id=44, pos=[(0,34,22)])
  sel.48: bits[34] = sel(uge.46, cases=[x2, literal.47], id=48, pos=[(0,35,25)])
  uge.51: bits[1] = uge(x4, literal.50, id=51, pos=[(0,36,42)])
  literal.52: bits[6] = literal(value=0, id=52, pos=[(0,36,57)])
  literal.59: bits[6] = literal(value=0, id=59, pos=[(0,39,49)])
  x12: token = after_all(__token, id=23)
  x29: bits[6] = dynamic_bit_slice(x5, literal.55, width=6, id=56, pos=[(0,37,22)])
  literal.67: bits[6] = literal(value=0, id=67, pos=[(0,42,50)])
  one_hot.31: bits[2] = one_hot(x17, lsb_prio=true, id=31, pos=[(0,28,23)])
  x23: bits[7] = one_hot(x14, lsb_prio=false, id=41, pos=[(0,32,25)])
  x27: (bits[33], bits[6], bits[6]) = array_index(x26, indices=[sel.48], id=49, pos=[(0,35,24)])
  sel.53: bits[6] = sel(uge.51, cases=[x4, literal.52], id=53, pos=[(0,36,34)])
  uge.60: bits[1] = uge(x7, literal.59, id=60, pos=[(0,39,42)])
  literal.61: bits[6] = literal(value=0, id=61, pos=[(0,39,57)])
  x20: (token, bits[47]) = receive(x12, predicate=x8, channel_id=0, id=34)
  uge.68: bits[1] = uge(x29, literal.67, id=68, pos=[(0,42,43)])
  literal.69: bits[6] = literal(value=0, id=69, pos=[(0,42,58)])
  encode.32: bits[1] = encode(one_hot.31, id=32)
  zero_ext.42: bits[6] = zero_ext(x8, new_bit_count=6, id=42)
  x11: bits[45] = literal(value=17592186044415, id=22, pos=[(0,21,23)])
  zero_ext.57: bits[45] = zero_ext(x23, new_bit_count=45, id=57)
  x28: (bits[33], bits[6], bits[6])[1] = array_update(x26, x27, indices=[sel.53], id=54, pos=[(0,36,28)])
  sel.62: bits[6] = sel(uge.60, cases=[x7, literal.61], id=62, pos=[(0,39,34)])
  zero_ext.64: bits[47] = zero_ext(x23, new_bit_count=47, id=64)
  x22: bits[47] = tuple_index(x20, index=1, id=39, pos=[(0,31,22)])
  sel.70: bits[6] = sel(uge.68, cases=[x29, literal.69], id=70, pos=[(0,42,34)])
  x16: token = after_all(x12, id=29)
  tuple_index.35: token = tuple_index(x20, index=0, id=35)
  x33: token = after_all(x12, id=66)
  literal.3: bits[1] = literal(value=1, id=3)
  literal.4: bits[1] = literal(value=1, id=4, pos=[(0,10,35)])
  literal.17: bits[32] = literal(value=0, id=17, pos=[(0,19,10)])
  literal.19: bits[32] = literal(value=1, id=19, pos=[(0,19,19)])
  x15: bits[1] = not(x6, id=28, pos=[(0,25,20)])
  x18: bits[1] = zero_ext(encode.32, new_bit_count=1, id=33, pos=[(0,28,23)])
  literal.36: bits[32] = literal(value=0, id=36, pos=[(0,30,25)])
  x21: token = tuple_index(x20, index=0, id=37, pos=[(0,30,24)])
  literal.38: bits[32] = literal(value=1, id=38, pos=[(0,31,23)])
  literal.40: bits[1] = literal(value=0, id=40, pos=[(0,32,36)])
  x24: bits[6] = sub(zero_ext.42, x10, id=43, pos=[(0,33,33)])
  x30: bits[1] = ule(x11, zero_ext.57, id=58, pos=[(0,38,26)])
  x31: (bits[33], bits[6], bits[6])[1] = array_update(x28, x27, indices=[sel.62], id=63, pos=[(0,39,28)])
  x32: bits[47] = and(zero_ext.64, x22, id=65, pos=[(0,40,36)])
  x34: (bits[33], bits[6], bits[6])[1] = array_update(x28, x27, indices=[sel.70], id=71, pos=[(0,42,28)])
  after_all.72: token = after_all(__token, x12, x16, tuple_index.35, x33, id=72)
  next (after_all.72, __state)
}

```

closed 2022-11-16 by copybara-service[bot]