google/xls #793: Gather codegen options into a proto

meheff, 2022-11-29:
The number of codegen options is growing. These are all passed in as flags which doesn't scale well. A better approach would be to define a proto which holds all the options. this proto can be passed as an input file to codegen_main. 

Not sure if we want to leave any of the options as flags on codegen_main. Maybe just a minimal simple set for doing commong things? I'm inclined to eliminate them all and have codegen_main only take a config proto. One downside is the proliferation of codegen option files. This could be mitigated by allowing the text proto to be passed in as a build rule attribute like so:

xls_ir_verilog(
    name = "foo",
    src = ":foo.ir",
    codegen_options = "{ generator=combinational }"
  ...
)

Might be a better way to encode protos in the build file, maybe via some starlark library.

vincent-mirian-google, 2022-11-29:
There is a [proto](https://github.com/google/xls/blob/main/xls/tools/codegen_flags.proto) defining the codegen flags.

We would need to consider 1 - readability (browsing to another file to see the options) and 2- a manner to overwrite the default values.

meheff, 2022-11-30:
Good point about the default values. I think these could be handled by either inverting bool flags (if default is now true) and/or changing the option to an enum with the default value being 0.

grebe, 2024-03-06:
We have a proto for codegen and scheduling flags now (see xls/tools/codegen_flags.proto). Would be nice to have for opt, but that's out of scope for this bug.

closed 2024-03-06 by grebe