google/xls #805: Sanitize or forbid DSLX NameDef or IR node names

RobSpringer, 2022-12-12:
In a recent module I wrote, I had an internal value (NameDef), called "output", that was preserved all the way to codegen as a wire...this made Verilog tools sad, as "output" is a reserved word. Either DSLX should forbid such words (WRONG ANSWER) or something downstream should sanitize them (CORRECT ANSWER).

cdleary, 2023-02-15:
I think we have sanitization in codegen now and can close, right @meheff ?

cdleary, 2023-02-15:
Or looks like our current `SanitizeIdentifier` in VAST doesn't particularly think about Verilog keywords, so maybe not done yet: https://github.com/google/xls/blob/87d78d495ca7e75ca83166791a018a0f7bdb6753/xls/codegen/vast.cc#L112

commit 80ba2810 (2025-05-23): Avoid name and keyword collisions in generated verilog.
Add a name uniquer on ModuleBuilder to prevent accidental collisions of identifiers in verilog modules (regs, wires, etc).

Other fixes, cleanups:

* xls::verilog::SanitizeIdentifier renamed to SanitizeVerilogIdentifier to distinguish it from other uses of sanitize.

* SanitizeVerilogIdentifier now sanitizes keywords.

* Raise an error if port names collide rather than silently renaming because port names are part of the interface and should not be changed.

* Don't call SanitizeVerilogIdentifer during block creation. We have a separate p [... cut]
PR #2229 (2025-05-23): Avoid name and keyword collisions in generated verilog.
Avoid name and keyword collisions in generated verilog.
Add a name uniquer on ModuleBuilder to prevent accidental collisions of identifiers in verilog modules (regs, wires, etc).

Other fixes, cleanups:

* xls::verilog::SanitizeIdentifier renamed to SanitizeVerilogIdentifier to distinguish it from other uses of sanitize.

* SanitizeVerilogIdentifier now sanitizes keywords.

* Raise an error if port names collide rather than silently renaming because port names are part of the interface and should not be changed.

* Don't call SanitizeVeri [... cut]
closed 2025-05-23 by copybara-service[bot]