google/xls #814: Optimizer requires long execution time (due to Z3 solver)

vincent-mirian-google, 2022-12-19:
The majority of the time is spent in running z3 solver in the mutual exclusion pass (Ref: `if (RunSolver(ctx, solvers::z3::BitVectorToBoolean(ctx, translated)) ==  Z3_L_FALSE)` at https://github.com/google/xls/blob/main/xls/passes/mutual_exclusion_pass.cc#L295).

A reduced IR input (the input to the pass):
```
package sample

file_number 0 "fake_file.x"

chan sample__x26(bits[2], id=0, kind=streaming, ops=receive_only, flow_control=ready_valid, metadata="""""")

top proc __sample__main_0_next(__token: token, __state: bits[1467], init={0x0}) {
  x2: bits[1467] = bit_slice_update(__state, __state, __state, id=4, pos=[(0,9,39)])
  literal.50: bits[1467] = literal(value=0x0, id=50)
  x8: bits[1] = xor_reduce(x2, id=10)
  x18: bits[1] = ne(x2, literal.50, id=51, pos=[(0,25,25)])
  x4: token = after_all(__token, id=6)
  x22: bits[1] = dynamic_bit_slice(x8, x18, width=1, id=24, pos=[(0,29,24)])
  x14: token = after_all(__token, x4, __token, id=16)
  x27: (token, bits[2]) = receive(__token, predicate=x22, channel_id=0, id=28)
  x17: token = after_all(__token, x14, x4, x14, id=19)
  x20: token = after_all(x14, __token, __token, x14, id=22)
  tuple_index.29: token = tuple_index(x27, index=0, id=29)
  after_all.34: token = after_all(__token, x4, x4, x14, x17, x20, tuple_index.29, id=34)
  next (after_all.34, x2)
}
```

vincent-mirian-google, 2022-12-19:
Another sample:
```
 package sample
 
 file_number 0 "fake_file.x"
 
 chan sample__x18(bits[27], id=0, kind=streaming, ops=receive_only, flow_control=ready_valid, metadata="""""")
 
 top proc __sample__main_0_next(__token: token, __state: bits[53], init={6004799503160661}) {
   x3: bits[53] = neg(__state, id=7, pos=[(0,10,18)])
   x4: bits[53] = or(__state, x3, id=84, pos=[(0,11,29)])
   x6: bits[53] = umul(x4, __state, id=12, pos=[(0,13,23)])
   x7: bits[53] = sub(x6, x3, id=13, pos=[(0,14,23)])
   x8: bits[1] = ule(x6, x7, id=14, pos=[(0,15,24)])
   x19: (token, bits[27]) = receive(__token, predicate=x8, channel_id=0, id=28)
   tuple_index.29: token = tuple_index(x19, index=0, id=29)
   after_all.51: token = after_all(__token, tuple_index.29, id=51)
   next (after_all.51, __state)
 }
```

vincent-mirian-google, 2022-12-19:
Another sample:
```
 package sample

 file_number 0 "fake_file.x"

 chan sample__x21(bits[103][13], id=0, kind=streaming, ops=receive_only, flow_control=ready_valid, metadata="""""")

 top proc __sample__main_0_next(__token: token, __state: bits[1784], init={0x7f_ffff_ffff_ffff_ffff_ffff_ffff_ffff_ffff_ffff_ffff_ffff_ffff_ffff_ffff_ffff_ffff_ffff_ffff_ffff_ffff_ffff_ffff_ffff_ffff_ffff_ffff_ffff_ffff_ffff_ffff_ffff_ffff_ffff_ffff_ffff_ffff_ffff_ffff_ffff_ffff_ffff_ffff_ffff_ffff_ffff_ffff_ffff_ffff_ffff_ffff_ffff_ffff_ffff_ffff_ffff_ffff_ffff_ffff_ffff_ffff_ffff_ffff_ffff_ffff_ffff_ffff_ffff_ffff_ffff_ffff_ffff_ffff_ffff_ffff_ffff_ffff_ffff_ffff_ffff_ffff_ffff_ffff_ffff_ffff_ffff_ffff_ffff_ffff_ffff_ffff_ffff_ffff_ffff_ffff_ffff_ffff_ffff_ffff_ffff_ffff_ffff_ffff_ffff_ffff_ffff_ffff_ffff_ffff_ffff_ffff_ffff}) {
   x2: bits[1261] = dynamic_bit_slice(__state, __state, width=1261, id=4, pos=[(0,11,27)])
   x4: bits[7] = dynamic_bit_slice(x2, x2, width=7, id=6, pos=[(0,13,21)])
   x6: bits[1261] = bit_slice_update(x2, x2, x2, id=8, pos=[(0,15,39)])
   x7: bits[7] = bit_slice_update(x4, __state, x2, id=9, pos=[(0,16,33)])
   bit_slice.70: bits[1254] = bit_slice(x6, start=7, width=1254, id=70, pos=[(0,15,39)])
   literal.67: bits[1254] = literal(value=0x0, id=67)
   x11: bits[7] = shll(x7, x2, id=18, pos=[(0,20,23)])
   bit_slice.72: bits[7] = bit_slice(x6, start=0, width=7, id=72, pos=[(0,15,39)])
   ne.75: bits[1] = ne(bit_slice.70, literal.67, id=75, pos=[(0,26,25)])
   ne.73: bits[1] = ne(x11, bit_slice.72, id=73, pos=[(0,26,25)])
   x14: token = after_all(__token, id=22)
   x17: bits[1] = or(ne.75, ne.73, id=74, pos=[(0,26,25)])
   x22: (token, bits[103][13]) = receive(x14, predicate=x17, channel_id=0, id=32)
   x23: token = tuple_index(x22, index=0, id=35, pos=[(0,30,24)])
   x28: token = after_all(x23, __token, id=45)
   after_all.46: token = after_all(__token, x14, x23, x28, id=46)
   next (after_all.46, __state)
 }
```

vincent-mirian-google, 2022-12-19:
Another sample:
```
package sample

 file_number 0 "fake_file.x"
 
 chan sample__x3(bits[985], id=0, kind=streaming, ops=receive_only, flow_control=ready_valid, metadata="""""")
 chan sample__x24(bits[20], id=1, kind=streaming, ops=receive_only, flow_control=ready_valid, metadata="""""")

 top proc __sample__main_0_next(__token: token, init={}) {
   x4: (token, bits[985]) = receive(__token, channel_id=0, id=5)
   x6: bits[985] = tuple_index(x4, index=1, id=10, pos=[(0,15,24)])
   x13: bits[354] = bit_slice(x6, start=361, width=354, id=18, pos=[(0,20,27)])
   x14: bits[354] = bit_slice_update(x13, x6, x13, id=19, pos=[(0,21,39)])
   literal.75: bits[354] = literal(value=0x3_ffff_ffff_ffff_ffff_ffff_ffff_ffff_ffff_ffff_ffff_ffff_ffff_ffff_ffff_ffff_ffff_ffff_ffff_ffff_ffff_ffaa_aaaa, id=75)
   x5: token = tuple_index(x4, index=0, id=8, pos=[(0,14,22)])
   x23: bits[1] = ne(x14, literal.75, id=78, pos=[(0,30,23)])
   x25: (token, bits[20]) = receive(x5, predicate=x23, channel_id=1, id=40)
   x9: token = after_all(__token, id=14)
   tuple_index.41: token = tuple_index(x25, index=0, id=41)
   after_all.49: token = after_all(__token, x5, x9, x9, tuple_index.41, id=49)
   next (after_all.49)
 }
```

vincent-mirian-google, 2022-12-21:
Another Sample:
```
package sample

 file_number 0 "fake_file.x"

 chan sample__x11(bits[6], id=0, kind=streaming, ops=receive_only, flow_control=ready_valid, metadata="""""")
 chan sample__x43(bits[32], id=1, kind=streaming, ops=receive_only, flow_control=ready_valid, metadata="""""")

 top proc __sample__main_0_next(__token: token, __state: bits[1881], init={0x0}) {
   x4: bits[56] = literal(value=1048576, id=17, pos=[(0,18,22)])
   x7: bits[56] = bit_slice_update(x4, __state, __state, id=21, pos=[(0,21,34)])
   literal.35: bits[56] = literal(value=24019198012642645, id=35, pos=[(0,30,10)])
   literal.26: bits[1] = literal(value=0, id=26, pos=[(0,25,49)])
   ne.142: bits[1] = ne(x7, literal.35, id=142)
   x12: (token, bits[6]) = receive(__token, predicate=literal.26, channel_id=0, id=27)
   x27: bits[6] = sign_ext(ne.142, new_bit_count=6, id=140)
   x14: bits[6] = tuple_index(x12, index=1, id=32, pos=[(0,27,21)])
   x28: bits[6] = bit_slice_update(x27, x14, x4, id=43, pos=[(0,33,34)])
   tuple_index.135: token = tuple_index(x12, index=0, id=135)
   x34: bits[1] = or_reduce(x28, id=60)
   x38: bits[1881] = neg(__state, id=66, pos=[(0,44,24)])
   after_all.136: token = after_all(tuple_index.135, __token, id=136)
   x42: bits[1] = shll(x34, x38, id=69, pos=[(0,47,26)])
   x44: (token, bits[32]) = receive(after_all.136, predicate=x42, channel_id=1, id=70)
   tuple_index.71: token = tuple_index(x44, index=0, id=71)
   x36: bits[1881] = not(__state, id=63, pos=[(0,42,24)])
   after_all.80: token = after_all(__token, tuple_index.135, tuple_index.71, id=80)
   next (after_all.80, x36)
 }
```

vincent-mirian-google, 2022-12-22:
```
 package sample

 file_number 0 "fake_file.x"

 chan sample__x2(bits[58], id=0, kind=streaming, ops=receive_only, flow_control=ready_valid, metadata="""""")
 chan sample__x18(bits[57], id=1, kind=streaming, ops=receive_only, flow_control=ready_valid, metadata="""""")

 top proc __sample__main_0_next(__token: token, __state: bits[1686], init={0x15_5555_5555_5555_5555_5555_5555_5555_5555_5555_5555_5555_5555_5555_5555_5555_5555_5555_5555_5555_5555_5555_5555_5555_5555_5555_5555_5555_5555_5555_5555_5555_5555_5555_5555_5555_5555_5555_5555_5555_5555_5555_5555_5555_5555_5555_5555_5555_5555_5555_5555_5555_5555_5555_5555_5555_5555_5555_5555_5555_5555_5555_5555_5555_5555_5555_5555_5555_5555_5555_5555_5555_5555_5555_5555_5555_5555_5555_5555_5555_5555_5555_5555_5555_5555_5555_5555_5555_5555_5555_5555_5555_5555_5555_5555_5555_5555_5555_5555_5555_5555_5555_5555_5555_5555_5555}) {
   x3: (token, bits[58]) = receive(__token, channel_id=0, id=141)
   x5: bits[58] = tuple_index(x3, index=1, id=10, pos=[(0,12,20)])
   x7: bits[58] = bit_slice_update(x5, __state, x5, id=12, pos=[(0,14,34)])
   x4: token = tuple_index(x3, index=0, id=8, pos=[(0,11,22)])
   x12: bits[1] = ugt(x7, x7, id=22, pos=[(0,19,25)])
   x19: (token, bits[57]) = receive(x4, predicate=x12, channel_id=1, id=36)
   tuple_index.37: token = tuple_index(x19, index=0, id=37)
   after_all.88: token = after_all(__token, x4, tuple_index.37, id=88)
   next (after_all.88, __state)
 }
```

vincent-mirian-google, 2022-12-22:
Another sample:
```
 package sample

 file_number 0 "fake_file.x"

 chan sample__x19(bits[8], id=0, kind=streaming, ops=receive_only, flow_control=ready_valid, metadata="""""")
 chan sample__x37(bits[43], id=1, kind=streaming, ops=receive_only, flow_control=ready_valid, metadata="""""")

 top proc __sample__main_0_next(__token: token, __state: bits[721], init={0x0}) {
   literal.6: bits[721] = literal(value=0xbe, id=6, pos=[(0,10,67)])
   ugt.115: bits[1] = ugt(__state, literal.6, id=115, pos=[(0,10,31)])
   sel.116: bits[721] = sel(ugt.115, cases=[__state, literal.6], id=116, pos=[(0,10,31)])
   x2: bits[721] = shrl(__state, sel.116, id=8, pos=[(0,10,27)])
   literal.140: bits[721] = literal(value=0xffff_ffff_ffff_ffff_ffff_ffff_ffff_ffff_ffff_ffff_ffff_ffff_ffff_ffff_ffff_ffff_ffff_ffff_ffff_ffff_ffff_ffff_ffff_ffff_ffff_ffff_ffff_ffff_ffff_ffff_ffff_ffff_ffff_ffff_ffff_ffff_ffff_ffff_ffff_ffff_ffff_ffff_ffff_ffff_ffff, id=140)
   ne.150: bits[1] = ne(x2, literal.140, id=150)
   x4: bits[722] = one_hot(x2, lsb_prio=false, id=21, pos=[(0,16,29)])
   sign_ext.152: bits[722] = sign_ext(ne.150, new_bit_count=722, id=152)
   x5: bits[722] = xor(x4, sign_ext.152, id=23, pos=[(0,17,27)])
   x6: bits[722] = bit_slice_update(x4, x5, __state, id=24, pos=[(0,18,38)])
   x9: bits[722] = bit_slice_update(x6, x6, __state, id=27, pos=[(0,21,38)])
   bit_slice.32: bits[4] = bit_slice(x9, start=0, width=4, id=32)
   x7: bits[4] = literal(value=7, id=25, pos=[(0,19,20)])
   x16__1: bits[1] = eq(bit_slice.32, x7, id=119, pos=[(0,25,25)])
   x20: (token, bits[8]) = receive(__token, predicate=x16__1, channel_id=0, id=38)
   x38: (token, bits[43], bits[1]) = receive(__token, channel_id=1, blocking=false, id=75)
   tuple_index.39: token = tuple_index(x20, index=0, id=39)
   x34: token = after_all(__token, id=71)
   tuple_index.76: token = tuple_index(x38, index=0, id=76)
   x8: bits[721] = not(x2, id=26, pos=[(0,20,22)])
   after_all.83: token = after_all(__token, tuple_index.39, x34, tuple_index.76, id=83)
   next (after_all.83, x8)
 }
```

meheff, 2023-02-28:
I believe these issues should be addressed with: 80d2bf3c74b0e6606853320e7ec3e0a5eb780488

closed 2023-02-28 by meheff