google/xls #824: DSLX interpreter should have a trace mode for procs

meheff, 2023-01-13:
It'd be good to have a mode in the interpreter where significant proc-related events are dumped to stderr. This could include:
* value sent on channel
* value received on channel
* proc tick started (and what the state is)
* proc tick ended
* etc
Would be good to expose this in dslx tests so you can do something like:

```
bazel run foo_test -- --test_filter=my_test_case --trace
```

cdleary, 2023-02-11:
Mark said he'd take this on as he had the best idea of what would be needed here.

meheff, 2023-02-27:
Tracing data sent on channels  added with e581b24b83111c56d2c9805093e6409599951ea3.

Still todo: tracing state at proc tick start/stop.

meheff, 2023-03-03:
Anther good feature would be to have some filtering on which channels/procs to trace, maybe based on name via regexp:

```
interpreter_main --trace_channels --trace_channels_filter=.*foo_chan.*
```

meheff, 2023-03-04:
A format preference should also be supported:

```
interpreter_main --trace_channels --trace_channels_format=x
```

Or something like that

meheff, 2023-03-21:
#906 filed separately for `trace_procs` feature.

grebe, 2023-09-06:
Can this be closed? It would be nice to have a flag that causes trace statements to be printed (right now, I think we need to to `--logtostderr --v=2`)

meheff, 2023-09-06:
I think all you need is --logtostderr to print the trace statements.  It is logged using XLS_LOG(INFO) by default.

In any case, this bug can probably be closed. Formatting the output of trace_channels is handled via `--format_preference`.

closed 2023-09-06 by meheff