google/xls #827: Support structs in trace_fmt

meheff, 2023-01-14:
only bits types work with trace_fmt. The following returns an error:

```
struct st0 {
  x: u1
}

fn foo() -> () {
  let _ = trace_fmt!("{}", st0 { x: false});
  ()
}
```

```
$ blaze-bin/third_party/xls/dslx/interpreter_main input.x
...
0044: 
0045: fn foo() -> () {
0046:   let _ = trace_fmt!("{}", st0 { x: false});
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^---------^ TypeInferenceError: st0 { x: uN[1] } trace_fmt! macro only supports printing simple bits values
0047:   ()
0048: }
```

meheff, 2023-01-17:
It would be good if field names were included in the output.

hongted, 2023-02-15:
Currently the ir converter will convert struct trace_fmt to ir such as the below

```
  _: token = trace(__token, __activated, format="st0{x: {}}", data_operands=[tuple_index.5], id=6)
```

This cannot be consumed by downstream tools such opt_main.  Error message given is
```
opt_main.cc:115] Check failed: ::absl::OkStatus() == (xls::tools::RealMain(positional_arguments[0])) (OK vs. INVALID_ARGUMENT: Invalid or unsupported format specifier "{x: {}" in format string "st0{x: {}}"; @ <unknown file>
```

Braces for the struct hierarchy should be escaped.  It should also remain escaped for the output of opt_main.
```
  _: token = trace(__token, __activated, format="st0{{x: {}}}", data_operands=[tuple_index.5], id=6)
```

commit b10a6120 (2023-02-16): Change struct formatting to use "{{" and "}}" to escape struct definitions.
Verilog output will un-escape these for proper display.

Fixes issue with ir ParseFormatString converting "{{" to "{" during parsing.

Fixes #827.

PiperOrigin-RevId: 509976755
closed 2023-02-16 by copybara-service[bot]