google/xls #836: Simplify DSLX proc syntax

meheff, 2023-01-14:
Example of curent syntax

```
pub proc foo {
  in_ch: chan<bool> in;
  out_ch: chan<bool> out;

  init { () }

  config (i: chan<bool> in, o: chan<bool> out) {
    let (a_in, a_out) = chan<bool>;
    spawn bar(a_in);
    spawn qux(a_out);
    (i, o)
  }
 
  next(tok: token, state: ()) {
    // use in_ch and out_ch
  }
}
```

There is a fair amount of replication and boiler plate in the config. One alternative might be to have the proc config arguments defined by the proc name kind of like a function. The config could even be folded up into the proc body. Maybe something like:

```
pub proc foo(in_ch: chan<bool> in, out_ch: chan<bool> out) {
  let a = chan<bool>;
  spawn bar(a.in);
  spawn qux(a.out);

  init { () }
 
  next(tok: token, state: ()) { 
    // use in_ch and out_ch
  }
}
```

meheff, 2023-02-06:
Another idea which maybe makes things a bit simpler and more explicit is to have an `run` statement (similar to `spawn` statement) which takes the initial state. With this the "body" of the proc is a number of statements which sets up the topology of connected procs (spawn statements) and then starts the running of the proc with an function-like invocation (run statement). 

```
pub proc foo(in_ch: chan<bool> in, out_ch: chan<bool> out) {
  let a = chan<bool>;
  spawn bar(a.in);
  spawn qux(a.out);
  run foo_body(state_t { ... });

  fn foo_body(tok: token, state: state_t) -> state_t { 
    // use in_ch and out_ch
  }
}
```

proppy, 2023-03-17:
Related #785

allight, 2026-11-14:
Closing this issue as fixed (accepted by @dplassgit): DSLX introduced simplified struct-style proc definitions (`proc Foo { ... } impl Foo { fn new(...) -> Self { ... } fn next(self) { ... } }`, represented by `ProcDef` in `xls/dslx/frontend/ast.h` and converted by `xls/dslx/fmt/legacy_proc_converter.cc`, e.g. `xls/examples/matmul_4x4/matmul_4x4.x:27-50`), eliminating separate `config`/`init` boilerplate. (Closed by Gemini)

closed 2026-11-14 by allight