google/xls #838: Print field names with trace! of a struct

meheff, 2023-01-17:
In DSLX, if trace! is called on a struct, the field names should be printed with the values.

cdleary, 2023-02-16:
@meheff do you know if this is solved with the `trace_fmt!` support? I think we still want to remove `trace!` in favor of `trace_fmt!` so there just one way to do it?

meheff, 2023-02-22:
I agree we should remove `trace!` and just have `trace_fmt!`.

I think one further enhancement would be to print field names on separate lines. For large structs, `trace_fmt!` print a super long line which is hard to read.

meheff, 2023-02-24:
1adcd99971c1d87b16418942878339267726a2e8 prints field names on different lines. This issue can be closed. I'll file a separate issue for removing trace!.

closed 2023-02-24 by meheff