google/xls #841: Add sub_with_carry builtin to XLS

meheff, 2023-01-19:
We have an add with carry builtin in XLS. It would be good to have a sub with carry one as well.

cdleary, 2023-03-01:
@meheff I think `add_with_carry` should probably become a stdlib function since there's no real need for it to be built-in. I guess it's unclear the optimizer will be able to make things more regular/uniform if we do a expand/add/slice today, though...

Thoughts on the motivation you had for the original `sub_with_carry` and whether a stdlib function would do what you were looking for?

meheff, 2023-03-01:
Having them be stdlib functions sounds fine to me.

Generally this issue was for a DSLX feature. The underlying IR and optimizations will be unaffected (of course) by the DSLX implementation details. I think that allowing adds to be mixed width (like multiplies) would make the underlying IR representation cleaner. Then add with carry would lower to an add with one extra bit of output. This does require signed add operations however.