google/xls #842: Second class functions

grebe, 2023-01-23:
It would be useful to be able to do something like

```
fn do_something<N:u32>(x: uN[N]) -> uN[N] { ... }

proc do_something_repeatedly<N: u32, function_to_do: Fn(uN[N])->uN[N] = do_something<N>> { ... }
```

It seems like first class functions are a pretty big change that breaks lots of abstractions, but second class (or whatever, I don't know what these classes all mean) functions would allow for a lot of proc reuse, especially if types can also be parameters to the proc.

cdleary, 2023-02-15:
If I'm understanding right, you're saying first class syntax to easily lift a function into a wrapping proc. After N extinguishes presumably it doesn't pop from the input channel ever again?

grebe, 2023-02-24:
I’m not sure exactly what you mean by N extinguishes- I’m not necessarily envisioning wrapping a function in a particular way. Maybe the proc invokes it multiple times and does other stuff. The idea is that in the proc context function_to_do is invocable. 

Maybe a more concrete example is a proc that reads from a set of channels, sorts the values, and sends them on in sorted order. It would be nice if you could pass in a comparison function to eg switch between ascending or descending order, or if type generics happen provide comparisons for complex types.

rmlarsen, 2024-12-04:
+1 for having this. It would be useful in avoiding code duplication, for example in testing:

```
import float32;

type F32 = float32::F32;

pub fn cosine(FP32: x) {
...
}

fn test_symmetry(FP32: x, ???: fn) {
  assert_eq(fn(x), fn(float32::negate(x)));
}

#[test]
fn test_cosine() {
  test_symmetry(float32::one(false), cosine)
  ...
}
```