google/xls #845: Support double colon in function/proc/block name in IR to support name spacing.

meheff, 2023-01-31:
A DSL proc named `bar` in a module foo produces an IR proc named `__foo__bar_0_next`. To help clean up readability and referring to the IR proc without name mangling (like in build rules), double colon should be supported in the IR for top-level identifiers. IR would look like:

```
proc foo::bar(__token: token, __state: (), init={()}) {
...
```

Also eliminating the `_0_next` suffix of the proc name would be good.

When lowering to verilog, these colons could be legalized by substituting colons with underscores or something, though typically a module name is specified explicitly  in codegen so this is not an issue.

grebe, 2023-02-16:
De-mangling parametrics, so that e.g. `foo:bar<3,4>` if `bar` is parameteric would be good too.

meheff, 2023-02-22:
I believe what LLVM does in these cases is use quoted strings as identifiers so you could have something like:

```
proc "foo::bar<3,4>"(...) {
```

We could do something similar.