google/xls #848: Imprecise error messages inside struct construction

taktoa, 2023-01-31:
If you construct a large struct and one of the fields of the struct has a parse error, you sometimes get an imprecise error pointing to the beginning of the struct construction. I've noticed this in two contexts: one where the parse error was an identifier that was not properly qualified (e.g.: it should have been `foo::bar` instead of `bar`) and another in which I attempted to do a bit slice with a constant rather than a literal (e.g.: `foo[bar:32]` instead of `foo[0:32]`).

cdleary, 2023-02-02:
Is there an internal bug # with a repro patch / client? (Otherwise will be hard to know when to close this out.)

meheff, 2023-02-11:
@cdleary see cl/508811001 which has a misconstructed const struct with an error message which only poinrts to the open brace. Various kinds of errors result in this same less than helpful message. Curiously I could not repro it with a simple struct like the one below which does give a good error message (!):

```
struct s {
  foo: u32,
  bar: bool,
}

const X = s {
  foo: u32:10,
  bar: BLAHBLAHfalse,
};
```

```
interpreter_main foo.x 
foo.x:8:8-8:21
0006: const X = s {
0007:   foo: u32:10,
0008:   bar: BLAHBLAHfalse,
~~~~~~~~~~~~~^-----------^ ParseError: Cannot find a definition for name: "BLAHBLAHfalse"
0009: };
0010: 
````

cdleary, 2023-02-14:
The reason why this happens (and only for colon-ref struct instantiation) is because we currently backtrack out of a struct instantiation failure. So then we have the colon-ref expression in our hands, and look ahead at the following token, and say "hey this `{` operator doesn't make sense here, you probably meant to just end your expression after that colon ref".

This could be solved with better diagnostic output, but I think the right way to approach it is to eliminate some of the existing need for backtracking in the parser. E.g. if parametric constant expressions had to be enclosed in braces like they do for parametric instantiations, I think that would let us remove a solid chunk of cases.

closed 2023-02-17 by copybara-service[bot]